OpenStickCommunity/GP2040-CE #965: Changes in custom layout for the display don't update in real-time

dangerousdonut, 2024-04-12:
### Expected Behavior

When going into the display configuration and choosing to show a custom button layout, you can move elements around (Start X, Start Y, etc.) and changes in these parameters would immediately be updated on the OLED display. 

### Current Behavior

Currently, in 0.7.8, making changes to the placement of a custom button layout isn't reflected until you save changes and reboot the device. However, in 0.7.7 changes in these parameters would still update in real time.

(Additionally, probably a separate issue, in 0.7.8 the button padding setting doesn't change the distance between buttons as it did in 0.7.7. It behaves unpredictably. For the left button layout, it seems to not work at all, while for the right layout it changes the vertical placement of the buttons.)

### Context

* Firmware version: 0.7.8 (GP2040-CE_0.7.8_Haute42.uf2)
* Board config name Haute 42 
* Name of device: Haute42 Hautepad P12
* Link to where you purchased the device: https://aliexpress.com/item/1005006780720770.html
* Operating System: Windows 10
* Browser: Firefox 124.0.2

### Steps to Reproduce

1. Update to 0.7.8 (Haute42 version)
2. In web-config, go into Configuration > Display Configuration
3. Under Layout Options, change Button Layout Left & Right to the Custom setting
4. Press B1 on the hardware to see button layout on the OLED
5. Changing the various custom parameters in web-config, like Star X, isn't reflected immediately on the screen
6. Additionally, the Button Padding option doesn't change distance between button icons

### Screenshots & Files

(If needed I can make a video, but I think the above covers the essentials.)

TheTrainGoes, 2024-04-21:
This is due to our re-write of the display back end.  

We will look to fix this for v0.7.9.

mikepparks, 2024-04-24:
Please confirm if the behavior still exists in PR #974.

dangerousdonut, 2024-04-25:
I'm not sure how I would test it 😅

TheTrainGoes, 2024-04-25:
Hi @dangerousdonut ,

You would need to go to the action for that PR, download the artifact, extract, nuke and try on your device.  

If thats not something you are comfortable doing then please wait until our new RC run before v0.7.9.

TheTrainGoes, 2024-06-03:
No answer from original poster, this is resolved in #974 from our testing.

closed 2024-06-03 by TheTrainGoes