OpenStickCommunity/GP2040-CE #973: When using the analog joystick of the Switch joycon, you can't push it full, only half. Would like to add a calibration option for the maximum value of the analog stick.

fatfatie, 2024-04-16:
When using the analog joystick of the Switch joycon, you can't push it full, only half. Would like to add a calibration option for the maximum value of the analog stick.
[image]

fatfatie, 2024-04-17:
[image]

TheTrainGoes, 2024-04-21:
@fatfatie can you let us know a bit more about your setup?   

What device you are using, how you have it wired and what your web-config setup is like.

fatfatie, 2024-04-22:
1 - Are you a vendor?
No

2 - What device are you using?
RP2040-Zero

3 - Where did you buy it? (post link to sales page)
https://www.waveshare.net/shop/RP2040-Zero.htm

4 - What version of the firmware are you currently running? (check this via web-config)
0.7.8

5 - Which distribution of the firmware are you currently running? (copy and paste the name of the firmware file you loaded on)
GP2040-CE_0.7.8_WaveshareZero.uf2

6 - What is the problem that you are having?
When using the analog joystick of the Switch joycon, you can't push it full, only half. Would like to add a calibration option for the maximum value of the analog stick.Using this analog joystick, the x-axis and y-axis voltages are between 0.7v-2.6v instead of 0v-3.3v, so the analog values are inaccurate.

7 - What trouble shooting steps have you already done?
The pin voltage was measured and the voltage value was not the standard 0v-3.3v.

fatfatie, 2024-04-22:
[image]

fatfatie, 2024-04-22:
[image]
[image]
[image]