OpenStickCommunity/GP2040-CE #976: Macro - Hold Press + Input results in flickering input rather than continuously held input

InfraredAces, 2024-04-17:
### Expected Behavior

After configuring the macro with a single line and combination of inputs, while holding the macro trigger, it holds continuously holds the input down until the trigger is released.

### Current Behavior

After configuring the macro with a single line and combination of inputs, while holding the macro trigger, the input will be held, but will occasionally flicker off

### Context

Firmware Version: 0.7.8
Board Config Name: GP2040-CE_0.7.8_Haute42COSMOX.uf2 
Device Name: S16
Operating System: 6.8.4-arch1-1 (64-bit)
Browser: Chromium

### Steps to Reproduce

1. Configure Macro to a pin as seen in the screenshot below
2. Press and hold button for macro
3. View input flickering using https://hardwaretester.com/gamepad

### Screenshots & Files

[image]

https://github.com/OpenStickCommunity/GP2040-CE/assets/44279698/a288a196-7feb-48bf-a609-219378c6605f

FeralAI, 2024-04-17:
@InfraredAces can you test out a build from #977 and see if that resolves the issue?

InfraredAces, 2024-04-17:
@FeralAI Using the build from #977 does seem to resolve the issue. Help the Macro for 10+ seconds and I did not find any flickering through the gamepad tester

closed 2024-05-10 by InfraredAces