OpenStickCommunity/GP2040-CE #981: GPIO pins used for internal board functions are still mapable in the pin mapping page

looper1234, 2024-04-19:
Problem
[image]
these pins are not available to end users 

but the pin mapping page lead me to believe pins such as 25 and 29 can be used for extra buttons

Solution

leave only the usable/customisable pins in the mapping page, and perhaps a brief reason why some pins are missing

TheTrainGoes, 2024-04-21:
@looper1234 This is heavily dependant on what device you are using.  If you are using a Pico those pins are reserved but if you are using a raw RP2040 you can have them used for other functions.  

@bsstephan will chat with you about this one.

TheTrainGoes, 2024-06-03:
This has been addressed and used pins are now block out on the configs that we support.

We cannot account for devices that are used outside of the supported ones.

closed 2024-06-03 by TheTrainGoes