OpenStickCommunity/GP2040-CE #995: 8BitDo Keyboard wired or fob support for Keyboard USB Host add-on

MukiEX, 2024-05-02:
### Problem

The 8BitDo Retro Mechanical Keyboard's keys don't appear to be detected by GP2040. Other keyboards tested on the same device (Logitech unifying adapter) worked just fine. Tested on two devices with the same behavior.

Requesting compatibility for the keyboard itself over cable or the 2.4GHz USB dongle it comes with.

If there's any info I can provide to assist with this via some kind of Linux USB dump command, please let me know. I'm not sure what the GP2040 software is expecting from a USB keyboard that the 8BitDo device isn't sending out.

MukiEX, 2024-05-02:
Actually, in retrospect, I should have probably at least picked up an lsusb/dmesg for this thing getting plugged in. The dongle alone seems to just dump a bunch of devices onto the system (a Raspberry Pi), including more than one keyboard:

```
[Thu May  2 00:58:59 2024] usb 1-1.1: USB disconnect, device number 3
[Thu May  2 00:59:02 2024] usb 1-1.1: new full-speed USB device number 6 using dwc_otg
[Thu May  2 00:59:02 2024] usb 1-1.1: New USB device found, idVendor=2dc8, idProduct=5201, bcdDevice= 1.00
[Thu May  2 00:59:02 2024] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu May  2 00:59:02 2024] usb 1-1.1: Product: 8BitDo Retro Keyboard
[Thu May  2 00:59:02 2024] usb 1-1.1: Manufacturer: 8BitDo
[Thu May  2 00:59:02 2024] input: 8BitDo 8BitDo Retro Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1:1.0/0003:2DC8:5201.0008/input/input29
[Thu May  2 00:59:02 2024] hid-generic 0003:2DC8:5201.0008: input,hidraw0: USB HID v1.11 Mouse [8BitDo 8BitDo Retro Keyboard] on usb-3f980000.usb-1.1/input0
[Thu May  2 00:59:02 2024] hid-generic 0003:2DC8:5201.0009: hiddev96,hidraw1: USB HID v1.11 Device [8BitDo 8BitDo Retro Keyboard] on usb-3f980000.usb-1.1/input1
[Thu May  2 00:59:02 2024] input: 8BitDo 8BitDo Retro Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1:1.2/0003:2DC8:5201.000A/input/input30
[Thu May  2 00:59:02 2024] hid-generic 0003:2DC8:5201.000A: input,hidraw2: USB HID v1.11 Keyboard [8BitDo 8BitDo Retro Keyboard] on usb-3f980000.usb-1.1/input2
```

From lsusb:

```
Bus 001 Device 006: ID 2dc8:5201 8BitDo 8BitDo Retro Keyboard
```

MukiEX, 2024-05-02:
Closing, as either keyboard host doesn't work in 0.7.8 (with keyboards that worked previously) or something happened that resulted in none of my previously-working keyboards working anymore. Not sure which of the two it is and I don't wanna risk wasting a moment of the team's time (I'm still 100% the 8bitdo keyboard doesn't work, but I can't test things as-is) so closing, barring further investigation on this request.

bsstephan, 2024-05-02:
I think it's an interesting request, and we do believe there are issues with the keyboard host in 0.7.8, so I'll go ahead and reopen this.

TheTrainGoes, 2024-05-02:
Thanks @bsstephan .

@MukiEX this is a known issue on v0.7.8 (keyboard host).

MukiEX, 2024-05-03:
Thank you both. Yeah, I didn't wanna continue the issue if I couldn't verify it, and it wasn't 'til late in the day that I noticed downgrading back to 0.7.7 resulted in the old keyboards working again. (8BitDo keyboard still did not)

TheTrainGoes, 2024-06-03:
This has been fixed in #1010

closed 2024-06-03 by TheTrainGoes