OpenStickCommunity/GP2040-CE #998: Issue with rotatory encoders

asturur, 2024-05-03:
Hi Everyone, i recently got the 0.78 update for a rotatory encoder and i noticed it does not work as expected with the encoder i had home.

### Expected Behavior

I can rotate the encoder and get a mapping on one axis like this

https://github.com/OpenStickCommunity/GP2040-CE/assets/1194048/8c656740-ca8c-4076-b15f-8064c3ce3a1c

### Current Behavior

The encoder reset to -1 after reaching +1

https://github.com/OpenStickCommunity/GP2040-CE/assets/1194048/b45c2169-c9d0-48be-b563-05b7bc2d3306

### Context

Rotatory encoder used: hw-40
[image]

And those are the specs:
https://www.elecom.sk/rotary-encoder-module-with-demo-code-for-uno-r3/

Encoder addon settings in 0.78
[image]

### Steps to Reproduce

Get a rotatry encoder of the same kind
Connect it and configure it as by example or in similar way, try to rotate it on both directions, one direction is flawed

mikepparks, 2024-05-03:
To add, since it was asked in the closed PR. This is what was used to develop the feature.

https://www.sparkfun.com/products/15141

It's fairly similar to what folks use anyway.

angelasvv, 2024-05-07:
May I interject?
I've noticed that whenever I enable a pin,that pin outputs a 3.3V, correct? And the GP28 pin is not working.
But the optical encoder I'm using has its own 5V on pin AB,will this affect the PICO? (NPN,600PPR)

PR #1004 (2024-05-07): Rotary encoder joystick mode overflows incorrectly
Fixed issue where rotary encoder would occasionally max out in the opposite direction in analog joystick modes.
Fixes #998 .
closed 2024-05-07 by arntsonl