OpenStickCommunity/GP2040-CE #1020: build instructions for developers

sgysh, 2024-05-22:
### Problem

I could not find any build instructions for developers.

### Solution

I could not find them in FAQ.
If they do not exist, it would be great if you could describe them in README.

### Do you plan to work on this feature yourself?

I'd like to try, but can't now because I don't know how to build.

TheTrainGoes, 2024-05-22:
Hi @sgysh ,

You can find some resources here:
https://gp2040-ce.info/development/firmware-development

closed 2024-05-22 by TheTrainGoes

sgysh, 2024-05-22:
Thank you very much, @TheTrainGoes .

What do you think of my idea of writing the link in README?
If it had been in README, I could have found it quickly without your help. I think it would be helpful to newcomers.

sgysh, 2024-05-22:
It's late at night here, so I'll think about it after waking up.