OpenStickCommunity/GP2040-CE #1030: Feature Request: Ability to disable "negative edge" for action buttons

pokeshark, 2024-05-29:
Ability to disable "negative edge" for action buttons (K1-K4, P1-P4).

For reference: negative edge is the act of releasing a button (instead of pressing it) to perform a [special move](https://glossary.infil.net/?t=Special%20Move). You might be surprised to learn that, in pretty much all fighting games, you can throw a [fireball](https://glossary.infil.net/?t=Fireball) by first pressing and holding a punch button by itself, then doing your [quarter circle](https://glossary.infil.net/?t=Quarter%20Circle) input, and finally releasing the button.

TheTrainGoes, 2024-05-29:
Hi @pokeshark ,

We have chatted internally and this is not something we are interested in implementing at this time.

closed 2024-05-29 by TheTrainGoes

bsstephan, 2024-05-29:
I will further clarify that the definition quoted above is describing *game* behavior, not *input device* behavior. We cannot control how a game registers a button press or hold.