OpenStickCommunity/GP2040-CE #1037: Missing labes for localization

rebornsick, 2024-06-04:
### Problem

There are some lables that needs to be translated.

#### SettingsPage

- sections on the left

[image]

- hotkey-actions.next-profile option seems not have a proper name yet.

[image]

#### DisplayConfig

- options of Flip Display

[image]

#### AddonsConfig

- options of On Board LED config

[image]

- options of Joystick Selection Slider

[image]

- options of DDI Dual D-Pad Mode and Combination Mode

[image]

- options of SOCD Slider Mode Default

[image]

- Lock Macro in Focus Mode Configuration

[image]

#### InputMacroAddon

- Macro name and Activation Type

[image]

### Solution

would love to see these lables added in the future.

and if we can set options like "D-Pad", "Digital", "Left Analog", "Right Analog" as universal lables since many selections have these same options.

### Do you plan to work on this feature yourself?

I'd love to but no, it's out of my ability.

SavageCore, 2024-06-04:
Thanks for this!

Lock Macro in Focus Mode Configuration is under Common.jsx `lock-macro`

See `Data/Addons.ts` for universal labels.

rebornsick, 2024-06-04:
I missed that one, so I need to wait for your PR to be merged before I can complete the localization. Then my PR can be merged, right? I’m not very clear about this process.

SavageCore, 2024-06-04:
Yes that will be ideal, there's more missing translations for us to go through.

commit ed341fa2 (2024-06-17): Add missing labels for localization (#1040)

* Fixes #1037

* Update these strings to match the "normal" way of doing things
closed 2024-06-17 by arntsonl