OpenStickCommunity/GP2040-CE #1044: Macro option "hold repeat" does not deactivate after release of button

wesleysni, 2024-06-04:
### Expected Behavior

While using the macro activation type option "hold repeat" while having macro button held down macro should repeat until it is released then it should stop replaying the macro sequence.

### Current Behavior

In current  v0.7.9 RC1 release, macro activates then does not stop, requiring to unplug the pcb.

### Context
* Firmware version =  v0.7.9 RC1
* Board config name = GP2040-CE_0.7.9_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2
* Name of device = GP-2040-CE v5.6E (usb-c-usb-b)
* purchased from Focusattack.com (currently unlisted/sold out)

### Steps to Reproduce

Screenshot provided will explain

### Screenshots & Files

[image]

TheTrainGoes, 2024-06-06:
Tested this on RC1.

The first test was a random setup of macros to check overall function - no issues.

The second test was a copy of the setup presented in the reported issue - was able to replicate.

The third test was changing `exclusive` to `interruptible` - no issues.

Conclusion from testing is that the combination of `exclusive` and `hold repeat` causes the loop.

arntsonl, 2024-06-07:
Should be fixed in RC2!

wesleysni, 2024-06-08:
Ok so with RC2 it does indeed stop upon releasing the button

problem is the macro is not executed on the initial button press, it is only fully executed if you hold the assigned button(s) for the whole duration of the macro. upon releasing the macro buttons at any time, the macro input sequence will stop at the very frame you release the macro button(s).

this is with hold repeat selected 

[image]

TheTrainGoes, 2024-06-08:
Hi @wesleysni ,

Good catch here.  We will have to talk internally to see if this is a bug or how the original function of toggle was intended to work.

TheTrainGoes, 2024-06-08:
Hi @wesleysni ,

We have talked internally and this is the intended function of the modes.  The same behaviour happens on the toggle setting.  

Our definitions for the various modes are:

- Press - Full press of button triggers the macro once (full string)
- Hold Repeat - Holding button to repeatedly triggers macro (will stop macro mid-string when released)
- Toggle - Full press of button causes the macro to repeatedly trigger, another full press of the button will stop the macro from repeatedly triggering (will stop macro mid-string when toggled off)

I will update the documentation to make this more clear.

closed 2024-06-09 by TheTrainGoes