OpenStickCommunity/GP2040-CE #1058: Add Haute42/COSMOX R16 16a button layout as a display layout

longxa762, 2024-06-13:
[image]

https://cosmoxgaming.com/hautepad/r16/

Latest haute42 board have a new layout for Stickless 16 that move the left thumb button to left pinky.

The firmare 0.7.9 update added the button layout to the "Custom LED Theme" page called "Stickless 16 A" but it's not available yet as a display layout.

TheTrainGoes, 2024-06-16:
Hi @longxa762 ,

To be clear, we do not add new stuff into the firmware 'just because' Haute42 has a new device out.

If there is enough demand for a layout we can look to get it added in.

secretanchitman, 2024-07-28:
Would like to second this request if possible! The R16 is becoming popular with it's unique layout and would love to see this added into the firmware to display correctly on the screen.

TheTrainGoes, 2024-07-30:
I've added the R16 display layout via #1097

closed 2024-08-11 by TheTrainGoes