OpenStickCommunity/GP2040-CE #1076: Xinput mode does not boot when connected to iPad

d4g, 2024-06-20:
### Expected Behavior

> Connect the haute pad to my iPad via USB-C in Xinput mode and it's registered as an xbox 360 controller

### Current Behavior

> The controller is not recognized. The green LED blinks.

### Context

> Please provide all relevant information about your setup. This is important in case the issue is not reproducible
> except under certain conditions. Important elements include:

* GP2040-CE_0.7.9_Haute42COSMOX.uf2 
* Haute42 COSMOX (GP2040-CE_0.7.9_Haute42COSMOX.uf2)
* Current: v0.7.9
* Latest: v0.7.9
* https://haute42.com/haute-board-mini/
* iPad OS 17.5.1
* Safari

### Steps to Reproduce

> Please provide detailed steps for reproducing the issue.
>
> 1. Press B Button and connect board to USB-C on iPad
> 2. Does not work

### Important
Other modes like PS4 mode or Switch Mode work.

TheTrainGoes, 2024-06-20:
Hey @d4g , generally I have found Xinput support on Apple stuff to be so-so.  I generally use Switch mode or Dinput (which will soon be HID-USB).

Is there a need for Xinput mode over any of the other modes that work as expected?

d4g, 2024-06-20:
No not really. I want it to use with Xbox Cloud Streaming and thought that xinput would probably the most compatible.

I found it interesting that even the controller signals something with the blinking led which does not occur when I connect it to PC.

TheTrainGoes, 2024-06-21:
I've tested on my iPad Pro on Xbox Cloud Streaming and there are multiple other modes that work just fine like PS4, Switch and keyboard. 

Going to close this as there is no action for us to take.

closed 2024-06-21 by TheTrainGoes

d4g, 2024-06-21:
I don't really get why you would close the issue. I get that's not of a high priority, but it's a bug nevertheless?

TheTrainGoes, 2024-06-21:
@d4g iOS is not an officially supported platform by us.  The official Xbox controller also does not work on iOS when connected via USB.  

Since there are multiple working modes for an unsupported platform I do not feel the need to keep this issue open and do not consider it to be a bug.

JulianAvilaP, 2025-04-01:
I agree with marking this as closed, but for future reference, iOS 18 added support for wired Xbox controllers for any users who might be interested in this issue

Ethan-Core2, 2026-01-07:
> I agree with marking this as closed, but for future reference, iOS 18 added support for wired Xbox controllers for any users who might be interested in this issue

Does this work over lighting, or just USB-C