OpenStickCommunity/GP2040-CE #1077: The 8-minute timeout issue using GP2040-CE_0.7.9_Haute42COSMOX.uf2 in PS5

joselitofilho, 2024-06-21:
### Expected Behavior

After the 8-minute timeout, my haute42 doesn't disconnect.

### Current Behavior

1. Plug your controller into the PS4/5
1. Press the PS button to initialize the controller
1. Play for roughly 8 minutes
1. Console rejects the controller due to failed authentication
1. User unplugs their non-functional controller
1. Go back to step 1 and repeat

### Context

* Firmware version: GP2040-CE_0.7.9_Haute42COSMOX.uf2
* Name of device: Haute 42, HautePad T16
* Link to where I purchased the device: https://www.amazon.com/dp/B0CN6M44F4?psc=1&ref=ppx_yo2ov_dt_b_product_details
* http://192.168.7.1/settings:
[image]
* http://192.168.7.1/settings:
[image]
[image]
[image]
[image]
[image]

### Steps to Reproduce

1. Plug haute42 on PS5
2. Wait 8 minutes

arntsonl, 2024-06-21:
Please read our documentation first on PS4 and PS5. You must have a supported dongle such as the Haute42 Booter 5 plugged into your device.

https://gp2040-ce.info/faq/faq-console-compatibility#playstation

closed 2024-06-21 by arntsonl

joselitofilho, 2024-06-21:
@arntsonl , how can I use my PS5 controller to authenticate my GP2040-CE device?

TheTrainGoes, 2024-06-21:
Removing this thread