OpenStickCommunity/GP2040-CE #1085: Add more macro button please!

WalkWithBlade, 2024-06-26:
Wish each macro button can be set as a key combination instead of corresponding to macro 1, macro 2...

bsstephan, 2024-06-26:
Can you please explain what problem you are encountering and attempting to solve with the feature request? We are not sure what you mean.

WalkWithBlade, 2024-06-26:
For example, since there is currently only one macro button that can set a combination key, move A can be released in both left and right directions. I have set it as a macro button+directional key. So move B cannot set combination keys anymore

WalkWithBlade, 2024-06-26:
> Can you please explain what problem you are encountering and attempting to solve with the feature request? We are not sure what you mean.

For example, since there is currently only one macro button that can set a combination key, move A can be released in both left and right directions. I have set it as a macro button+directional key. So move B cannot set combination keys anymore

arntsonl, 2024-06-27:
This is not something we are looking to implement. Unique, per-move macro assignment that does not follow any other macro layouts out there will have to be maintained by another group as this is too specific and custom for our firmware.

closed 2024-06-27 by arntsonl