OpenStickCommunity/GP2040-CE #1092: [BUG] Function hotkeys broken once function key enabled

Sean-Paulk, 2024-07-20:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on firmware capabilities and usage.
> * Make sure that your issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).
> * Confirm your issue exists on the latest version of the firmware, found
  [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases).

### Expected Behavior

> Please describe the behaviour you are expecting.
1. Function hotkeys are disabled when no function key is set.
2. Afterwards, hotkeys are usuable but only if enabled (right position for the slider).
3. Function hotkeys should not activite when in the off position (left position).
4. Single button hotkeys are still usuable outside of the hotkey e.g. : “Fn + Down  = Dpad Digital” doesn’t impiede the usage of using down as assigned from the pin mapping section.

### Current Behavior

> What is the current behaviour? If this is something that worked in the past, be sure to include the conditions in
> which it did (prior version of the firmware, a different GP2040-CE device where it works, etc.).

After setting a dedicated function key, the function hotkeys are enabled even if disabled (left position).
In addition, “Fn + Down  = Dpad Digital” doesn’t allow the down button to no longer function.
From memory, before upgrading this was not a problem.

### Context

> Please provide all relevant information about your setup. This is important in case the issue is not reproducible
> except under certain conditions. Important elements include:
>
> * Firmware version (please get the full version listed in webconfig, if at all possible)
> * Board config name (also from webconfig, please)
> * Name of device
> * Link to where you purchased the device
>
> If your issue is with the web configurator, also include:
>
> * Operating System
> * Browser

Version 
RP2040 Advanced Breakout Board - USB Passthrough (GP2040-CE_0.7.9_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2)
Current: v0.7.9

### Steps to Reproduce

> Please provide detailed steps for reproducing the issue.
>
> 1. Step 1
> 2. Step 2
> 3. ETC...

### Screenshots & Files

> Please include any relevant screenshots, photos/recordings, or files here.

TheTrainGoes, 2024-07-20:
Hi @Sean-Paulk ,

This is not a bug. The Function pin is optional to assign.  For those without a Function pin assigned the hotkeys will still work with the assigned key combos.   

For those with a Function pin assigned you can make the key combinations significantly simpler or have multiple layers or shortcuts.  

Closing.

Sean-Paulk, 2024-07-20:
Correct me if I am wrong but this below means off.
[image]

And this below means on?

[image]

TheTrainGoes, 2024-07-22:
Hi @Sean-Paulk ,

This is a slider for the `NF` portion of the hotkey only, not the entire sequence.  

In example `1` above, if you have `Start` + `Select` + `P1` set to `Home` then there is no need to press the `FN` button and pressing `Start` + `Select` + `P1` will press the `Home` button.

In example `2` above, if you have `Start` + `Select` + `P1` set to `Home` then you will need to press the `FN` button and pressing `Start` + `Select` + `P1` which, when all pressed together, will press the `Home` button.

Sean-Paulk, 2024-07-22:
@ TheTrainGoes,

> Hi @Sean-Paulk ,
>
> This is a slider for the `NF` portion of the hotkey only, not the entire sequence.
>
> In example `1` above, if you have `Start` + `Select` + `P1` set to `Home` then there is no need [...]

I did not know this. This would explain why `Start` + `Down` = `Home` in situation 1. would disable the `Down` button from registering. I don't think this nuanced information is referenced in the documentation.

TheTrainGoes, 2024-07-22:
@Sean-Paulk please check over our documentation here for a writeup on how to use the FN feature.  

https://gp2040-ce.info/web-configurator/menu-pages/settings

If you think there is something that could be more clear please feel free to submit a PR with your ideal changes and we can review it.

Sean-Paulk, 2024-07-23:
That is a great idea. I can't find the location of the page to edit. I see [README.md](https://github.com/OpenStickCommunity/GP2040-CE/blob/main/README.md) under main directory and [ddi-socd.md](https://github.com/OpenStickCommunity/GP2040-CE/blob/main/docs/ddi-socd.md) under the docs directory but not the function or the other pages available at [https://gp2040-ce.info](https://gp2040-ce.info).

Sean-Paulk, 2024-07-23:
@TheTrainGoes 

That is a great idea. I can't find the location of the page to edit. I see [README.md](https://github.com/OpenStickCommunity/GP2040-CE/blob/main/README.md) under main directory and [ddi-socd.md](https://github.com/OpenStickCommunity/GP2040-CE/blob/main/docs/ddi-socd.md) under the docs directory but not the function or the other pages available at [https://gp2040-ce.info](https://gp2040-ce.info).

TheTrainGoes, 2024-07-24:
@Sean-Paulk ,

This is all on the `sites` section of the repo:

https://github.com/OpenStickCommunity/Site/blob/main/docs/web-configurator/menu-pages/01-settings.mdx#hotkey-settings