OpenStickCommunity/GP2040-CE #1106: How keyboard macro programming works

rmamba, 2024-08-09:
I can't figure out how to program double clicks in keyboard macros.

For example I'm having trouble programming 
LEFT(on)
DOWN(on)DOWN(off)
DOWN(on)DOWN(off)
LEFT{off)

so need to press down twice while holding left pressed.

Looked at the documentation, but there is very little explenation on how macro programming works.

TheTrainGoes, 2024-08-09:
Hey @rmamba ,

GP2040-CE will look at each line individually, so to have left held and have it press down twice while holding left you would want something like this:

LEFT(on)
LEFT(on) DOWN(on)
LEFT(on) DOWN(off)
LEFT(on) DOWN(on)
LEFT(on) DOWN(off)
LEFT{off)

rmamba, 2024-08-09:
Does it automatically figure out on/off?

I don't really get the logic in these form, but this is what you are saying @TheTrainGoes right?
[image]

rmamba, 2024-08-10:
FYI,

It works, replaced last entry with None only.

closed 2024-08-11 by TheTrainGoes