OpenStickCommunity/GP2040-CE #1127: Unable to map keyboard "Escape" key to Function button.

skdursh, 2024-09-01:
A few months ago I purchased a Haute Pro Stickless (16+5 button layout). It's been working great so far and I have used the web configurator numerous times without issue. Recently I purchased Street Fighter 6 and for whatever reason there is no way in-game to remap the Escape key (which is used for opening several important menus) to anything else, so I am attempting to set up an SF6 profile on the stickless with the configurator to have the Function key act as a press of Escape. I currently have Fn mapped to Pin 26 and in the Keyboard Mapping section of the configurator there is an explict option to map the Fn key to any other standard keyboard key of my choosing, but whenever I try to do so (and then save it at the bottom), when I reboot the device to try it out it automatically unmaps itself. Is there a reason for this? If there is, then why is there even the option to remap the Fn key in the Keyboard Mapping section to begin with? Is there any way to fix it? I would prefer not to remap any of the other Option keys as I have them set up in a specific manner for Fightcade and also don't actually use the Fn key for anything else right now as I don't really have any need for any of the options offer in the Function menu for the time being so I really would just prefer to remap it as specified above if possible.

Additional Info:
Stickless - Haute 42 Haute Pro 16+5 Stickless
OS - Windows 10
Browser - Current version of Ungoogled-Chromium (in all the ways that actually matter it has version parity with and functions exactly the same as a standard installation of Chromium, but with Google's spyware removed from it)

TheTrainGoes, 2024-09-01:
@skdursh what mode are you using the controller in?

skdursh, 2024-09-01:
> @skdursh what mode are you using the controller in?

Keyboard mode.

TheTrainGoes, 2024-09-02:
@skdursh can you please attach a backup of your web-config so that I can load it onto my T16.  

Also, what is a 16+5 stickless?

TheTrainGoes, 2024-09-02:
Also @skdursh what version of the firmware are you using?

bsstephan, 2024-09-17:
I think the ability to map Fn in the keyboard mapping is an error of older firmware, BTW. The Fn key is not meant to be sent out to the host, rather it is a button that is only used by the internal logic for chord-like hotkey behavior. Checking main, it is no longer possible to assign Fn to a keyboard key, and instead there are new extra buttons that you might be able to use instead. I'm going to close this and hopefully 0.7.10 solves your use case.

closed 2024-09-17 by bsstephan