OpenStickCommunity/GP2040-CE #1134: 0.7.9 oled display issue

1142694044, 2024-09-14:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on hardware requirements and usage.
> * Make sure that your issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).

### Expected Behavior

>latest 0.7.9 oled display properly

### Current Behavior

>when i upgrade 0.77 to 0.79 fireware.oled display only respond right layout section(all layout.eg stick stickless).left section for direction not showing responded on oled screen.
when i downgrade to 0.7.7 exactly same setting up. problem disappeared.display and operation normal

### Context

>mayflash installed inside usb port:
i2c0 0x3c sda 0 scl 1
>
> rp2040ce
> aliexpress
>.0.7.9

> mayflash 1.1 insstall inside for ps5

> pico

### Steps to Reproduce

> Please provide detailed steps for reproducing the issue.
>
> 1. Step 1 nude upgrade to 0.7.9
> 2. Step 2 display error on left direction section
> 3. downgrade to 0.7.9 everything performance normal

### Screenshots & Files

> Please include any relevant screenshots or files here, including photos of the front and back of the board.

TheTrainGoes, 2024-09-14:
Hi @1142694044 ,

it is very hard to follow what you have written.  Please write it out again clearly and include pictures of the issue.

1142694044, 2024-09-14:
Upgrade 0.77 to 0.79 fireware.oled display only respond right layout section(all layout.eg stick stickless).left section for direction not showing responded on oled screen.
when i downgrade to 0.7.7 exactly same setting up. problem disappeared.display and operation normal
Context

mayflash installed inside usb port:
i2c0 0x3c sda 0 scl 1

rp2040ce
aliexpress
.0.7.9
mayflash 1.1 insstall inside for ps5
pico
Steps to Reproduce

Please provide detailed steps for reproducing the issue.

Step 1 nude upgrade to 0.7.9
Step 2 display error on left direction section
downgrade to 0.7.9 everything performance normal
Screenshots & Files

Please include any relevant screenshots or files here, including photos of the front and back of the board.
 @[1142694044](https://github.com/1142694044) 1142694044 added the [bug](https://github.com/OpenStickCommunity/GP2040-CE/labels/bug)  label [10 hours ago](https://github.com/OpenStickCommunity/GP2040-CE/issues/1134#event-14256404950)
@TheTrainGoes
Contributor
[TheTrainGoes](https://github.com/TheTrainGoes) commented [9 hours ago](https://github.com/OpenStickCommunity/GP2040-CE/issues/1134#issuecomment-2350802167)
Hi @1142694044 ,

it is very hard to follow what you have written. Please write it out again clearly and include pictures of the issue.
 @1142694044

Add a comment

Comment
 Add your comment here...
 
Remember, contributions to this repository should follow our [GitHub Community Guidelines](https://docs.github.com/articles/github-community-guidelines).
Assignees
No one assigned
Labels
[bug](https://github.com/OpenStickCommunity/GP2040-CE/labels/bug)
Projects
None yet
Milestone
No milestone
Development
No branches or pull requests

Notifications
Customize
You’re receiving notifications because you were mentioned.
2 participants

mikepparks, 2024-09-22:
Sounds to me like you unintentionally switched to Left Analog input and the D-Pad button inputs are not displaying. [Hotkeys](https://gp2040-ce.info/hotkeys#dpad-digital)