OpenStickCommunity/GP2040-CE #1135: Diagonal Direction Inputs for Macro dropdown menu

Helder1981, 2024-09-14:
### Problem

> Which problem is this feature request solving?
> add diagonal inputs for Macros
> Provide a clear and concise description of what the problem is. For example, "I'm always frustrated when..."

### Solution

> Describe the solution you would like to see
>Add Diagonal inputs in the drop down menu instead of just the 4 cardinal directions, this would make 1 button special moves much easier to perform. 

### Do you plan to work on this feature yourself?
No

> Beyond a simple yes or no, we are also willing to help with partial solutions or to refine a proof of concept, if you
> are able to contribute.
Yes

TheTrainGoes, 2024-09-14:
Hi @Helder1981 , 

Entering both directions on the same line should address this.  

Are you having an issue with this approach?

Helder1981, 2024-09-14:
> Hi @Helder1981 ,
> 
> Entering both directions on the same line should address this.
> 
> Are you having an issue with this approach?

No I don't have the macro gpio on my controller so I could not test it but I found it weird when looking through the menus that no Diagonals were there but what you said makes sense now, thank you.

TheTrainGoes, 2024-09-14:
We will also have multiple inputs per pin (main mapping, not macros) which will allow you to have dedicated inputs like `up` + `right` on one input.  

This will be included in the next release.

closed 2024-09-14 by TheTrainGoes