OpenStickCommunity/GP2040-CE #1142: join more buttom for keyboard

nibsen1985, 2024-09-19:
We use this set for doing paper work.Please give us more bottom like insert and Ctr+ v for website setting ,thanks

nibsen1985, 2024-09-25:
I am game player and also a stock trader.My staffs  and I found the hitbox is better than normal keyboard . “Insert”important button  for stock trader .but some trader  love some shortcut keys like “ctr +v” or “alt +a”

---- Replied Message ----
| From | ***@***.***> |
| Date | 09/25/2024 22:00 |
| To | OpenStickCommunity/GP2040-CE ***@***.***> |
| Cc | nibsen1985 ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [OpenStickCommunity/GP2040-CE] join more buttom for keyboard (Issue #1142) |

Hi @nibsen1985 ,

Just to ensure I am following, you are asking for two separate things here:

1 - You would like the insert button added for keyboard mode
2 - You would like the ability to map two buttons in keyboard mode - example Control + V

nibsen1985, 2024-09-25:
Yes you are right

---- Replied Message ----
| From | ***@***.***> |
| Date | 09/25/2024 22:00 |
| To | OpenStickCommunity/GP2040-CE ***@***.***> |
| Cc | nibsen1985 ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [OpenStickCommunity/GP2040-CE] join more buttom for keyboard (Issue #1142) |

Hi @nibsen1985 ,

Just to ensure I am following, you are asking for two separate things here:

1 - You would like the insert button added for keyboard mode
2 - You would like the ability to map two buttons in keyboard mode - example Control + V

ez2torta, 2024-09-29:
@nibsen1985 i'm super curious on how you use the hitbox for trading, i don't do trading myself but i´d like to know other uses for controllers.
can you share your layout for buttons? 
Thanks!

nibsen1985, 2024-09-29:
I am using keyboard for work . But keyboard’s buttons let the movement slowly than hitbox.So I diy it (silly lol).But it still can’t reach the standard I want .
I get the new hitbox .I can drag any key I want in the website .but it is so damn heavy .

​

---- Replied Message ----
| From | Paulo ***@***.***> |
| Date | 09/29/2024 23:00 |
| To | OpenStickCommunity/GP2040-CE ***@***.***> |
| Cc | nibsen1985 ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [OpenStickCommunity/GP2040-CE] join more buttom for keyboard (Issue #1142) |

@nibsen1985 i'm super curious on how you use the hitbox for trading, i don't do trading myself but i´d like to know other uses for controllers.
can you share your layout for buttons?
Thanks!