OpenStickCommunity/GP2040-CE #1143: Suggestion: Togglable Custom Splash Screen

teddydoree, 2024-09-20:
### Problem

I love having my custom splash screen on display, but setting the splash duration to 0 means I can never see the info screen without entering the Web Configurator.

### Solution

I suggest adding a Hotkey Setting to toggle between the custom Splash Screen and the button info screen. Another solution could be including an option to reverse the splash duration, so that on boot the Info screen would show for an amount of time then switch to the custom Splash Screen.

### Do you plan to work on this feature yourself?

Never coded firmware before, but open to trying if the learning curve isn't too steep.

TheTrainGoes, 2024-09-29:
Hi @teddydoree ,

You are welcome to work on this one yourself if you like.  

I would look at the code behind the optinon to have the splash display for X seconds.

TheTrainGoes, 2024-12-28:
We are going to close this as not planned.

closed 2024-12-28 by TheTrainGoes