OpenStickCommunity/GP2040-CE #1145: Compatibility rasbery pico pi 1 and pico pi 2 ?

InfernalWolf666, 2024-09-21:
I decided to buy the second version of pico pi 2. Are the firmware compatible?

Sorry for the stupid question, I only used arduino uno.

InfernalWolf666, 2024-09-21:
[image]

mikepparks, 2024-09-21:
Not currently supported.  We're investigating the next steps, but they're two different chips.

closed 2024-09-21 by mikepparks

InfernalWolf666, 2024-09-21:
> В настоящее время не поддерживается. Мы изучаем следующие шаги, но это два разных чипа.

I'm trying to find information, they seem to be fully compatible. Either an ARM processor can work if the code is written for it, or a RISK processor will work if the code is written for it. They don't work at the same time. But I still haven't figured it out, there's not enough information.

mikepparks, 2024-09-21:
It is not compatible with the project at the current time.  We will need to make modifications to make sure that everything works as expected. It is planned, but nothing to announce yet.