OpenStickCommunity/GP2040-CE #1155: Profile 2+ Pin Mapping is Wiped on Reboot or Does not Save

brentaro, 2024-09-27:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on firmware capabilities and usage.
> * Make sure that your issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).
> * Confirm your issue exists on the latest version of the firmware, found
  [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases).

### Expected Behavior

Create profile 2 and retaining all pin mappings upon reboot.

### Current Behavior

After creating a profile 2 and making changes or even copying base profile, hitting save and rebooting into controller. ALL pin mappings on Profile 2 are unset/wiped. If you ever switch to Profile 2 in controller mode, none of the buttons/pins are mapped and you're forced to manually go into BOOTSEL and use the force webconfig flash file to get back into web config. Same behavior on additional profiles (profile 3, etc.)

### Context

Firmware Version: v0.7.10-RC1
Board config name: GP2040-CE_0.7.10_Haute42COSMOX.uf2
Name of device: Haute T16

### Steps to Reproduce

1. Navigate to pin mapping
2. Add Profile (2)
3. Copy Base Profile and/or Map Any pins manually
4. Hit Save
5. Reboot into Web Config or controller then boot up into web config
(Tested with GP2040 and PS4 input display in web config)

### Screenshots & Files
[image]

> Please include any relevant screenshots, photos/recordings, or files here.

brentaro, 2024-09-27:
Also to note, there's doesn't seem to be a way to delete profiles. Only Disable them.

bsstephan, 2024-09-28:
This should be #1154.

bsstephan, 2024-09-30:
This will be fixed in v0.7.10-RC2.