OpenStickCommunity/GP2040-CE #1170: Mapping of display screen.

d-411, 2024-09-30:
[image]
### Problem

Dear author, I have modified the mapping of the keys. However, the key mapping on the screen is still the original one. Is there a way to modify the mapping of the screen?

d-411, 2024-09-30:
My option is i2c0 and the version is 7.10，button layout is stickless.

d-411, 2024-09-30:
My option is i2c0 and the version is 7.10，button layout is stickless.don't pay attention to the feedback of LED lights. Please focus on the key mapping echo of LED displays.

mikepparks, 2024-09-30:
Changing the display layout buttons is currently not possible without a custom build. This is planned for a later version.

closed 2024-09-30 by d-411