OpenStickCommunity/GP2040-CE #1183: enabling analog addon bricks the controller after reconnect/reset

MegaBitmap, 2024-10-03:
### Expected Behavior

The controller should still work after multiple disconnect reconnect cycles with the analog addon enabled.

### Current Behavior

With the current release 0.7.10, enabling the analog addon prevents the controller from working after a disconnect or reset.

### Context

There seems to be a problem with this commit:
https://github.com/OpenStickCommunity/GP2040-CE/commit/7491db64993945c66688a05ffd2602e3e3b522a1

With the current release 0.7.10, enabling the analog addon prevents the controller from working after a disconnect or reset.
I used the force_webconfig.uf2 to enter the web config then disabled the analog addon.
This fixed the controller buttons but then the analog sticks are disabled.
If I re-enable the analog sticks they work initially, but fails after a controller disconnect or reset.
I tried the PXPGamepad uf2 as well as WaveshareZero uf2 and both have the same issue.
The controller is not detected (I tried xinput and switch mode) and the only way to fix it was to use boot select mode.
The controller works if rebooted from within the webconfig gui, but as soon as the controller is disconnected or reset it fails.

I made a testing branch with this commit reverted:
https://github.com/MegaBitmap/GP2040-CE/tree/testing-24-10-3
After compiling a uf2 and flashing it, everything works perfectly with the analog addon enabled.
Even after multiple disconnect and reconnect/reset cycles everything works fine.

(Unrelated issue) Also with the latest pico-sdk there is a build error conflicting with Pico-PIO-USB.
I switched to pico-sdk v1.5.1 and there is no longer a build error with pio_sm_set_jmp_pin

Using windows 11 23H2 with edge browser

### Steps to Reproduce

1. enable analog addon in firmware 0.7.10
2. Initially (on the fist reboot from within web config) everything works, analog sticks and all.
3. After a USB disconnect or controller reset, after plugging the controller back in, the controller is not detected.
4. The only way to fix it was to use the boot select button and flash old firmware or use  force_webconfig.uf2.

### Screenshots & Files

Reverting the changes from analog.cpp and analog.h fixes the issue.

closed 2024-10-04 by arntsonl

MegaBitmap, 2024-10-04:
Thank you for solving the issue. That was fast!

arntsonl, 2024-10-04:
Hot fix is live!