OpenStickCommunity/GP2040-CE #1187: Allow SOCD settings to be stored alongside a Profile

j-zhao, 2024-10-04:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on firmware capabilities and usage.
> * Make sure that your idea has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).
> * Confirm the feature does not exist in the latest version of the firmware, found
  [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases).

### Problem

Different games have different optimal SOCD resolution (e.g. neutral for SF6, vs SOCD-U for tag fighters). Often times, this warrants changing a profile as well as inputting a shortcut on the device to change SOCD. 

### Solution

Store SOCD settings alongside a profile. This allows a few improvements:
- Reducing one less device input to change SOCD alongside changing profiles
  - Intuitively, this makes it very easy to set-it-and-forget-it for each game
  - With the recent 0.7.10 release, custom profile names will benefit displays. Seeing the correct profile on your device display is insurance that your SOCD is set correctly, instead of managing two changes on first-plug.

### Do you plan to work on this feature yourself?

Yes. I will create a PR to kick things off, and will engage w/ the community and Discord to get this merged.

### Notes
- [Discord Thread](https://discord.com/channels/1049366310389289001/1049370387563163731/1291809976909693069)

bsstephan, 2024-10-04:
As a breadcrumb you will be looking at adding something new to ProfileOptions in the protobuf structure. I don't have an opinion currently on if that should be more GamepadOptions (most fields of which would be ignored in the profiles) or a new structure (which might suggest GamepadOptions needs a refactoring) but be aware that is a critical element of how to do this.

febs, 2024-12-06:
Would it be feasible to also save the rgb led profile as well? Thank you for reading.

TheTrainGoes, 2026-11-05:
Implemented in #1803

closed 2026-11-05 by TheTrainGoes