OpenStickCommunity/GP2040-CE #1193: WebConfig broken in pull request (#1179).

darkxex, 2024-10-10:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on hardware requirements and usage.
> * Make sure that your issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).

### Expected Behavior

web config open without problems.

### Current Behavior

in this pull request #1179 merge into main, the web config does not open.
https://github.com/OpenStickCommunity/GP2040-CE/commit/9e5c338bcac1f0abcae4ce84c9a1b47da60f4790

### Context

Tested in Raspberry Pico W.

### Steps to Reproduce

> Please provide detailed steps for reproducing the issue.
>
Just install the uf2.

arntsonl, 2024-10-10:
We are aware that Pico SDK 2.0 broke web config, thank you. It will be functional once we work out why Pico SDK 2.0 does not play well with the RNDIS library.

TheTrainGoes, 2025-01-03:
Going to close this as issue will be resolved if we get it working with SDK 2.x or if we decide to just keep at 1.5x

closed 2025-01-03 by TheTrainGoes