OpenStickCommunity/GP2040-CE #1199: Building error

chiyu521, 2024-10-18:
[ 10%] Building C object lib/PicoPeripherals/CMakeFiles/PicoPeripherals.dir/__/pico_pio_usb/src/pio_usb.c.obj
In file included from /home/chiyu/pico/GP2040-CE/lib/pico_pio_usb/src/pio_usb.c:24:
/home/chiyu/pico/GP2040-CE/lib/pico_pio_usb/src/usb_rx.pio.h:186:29: error: redefinition of 'pio_sm_set_jmp_pin'
  186 | static __always_inline void pio_sm_set_jmp_pin(PIO pio, uint sm, uint jmp_pin) {
      |                             ^~~~~~~~~~~~~~~~~~
In file included from /home/chiyu/pico/GP2040-CE/lib/pico_pio_usb/src/pio_usb.c:14:
/home/chiyu/pico/pico-sdk/src/rp2_common/hardware_pio/include/hardware/pio.h:1450:20: note: previous definition of 'pio_sm_set_jmp_pin' with type 'void(pio_hw_t *, uint,  uint)' {aka 'void(pio_hw_t *, unsigned int,  unsigned int)'}
 1450 | static inline void pio_sm_set_jmp_pin(PIO pio, uint sm, uint pin) {
      |                    ^~~~~~~~~~~~~~~~~~
make[2]: *** [lib/PicoPeripherals/CMakeFiles/PicoPeripherals.dir/build.make:905：lib/PicoPeripherals/CMakeFiles/PicoPeripherals.dir/__/pico_pio_usb/src/pio_usb.c.obj] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2645：lib/PicoPeripherals/CMakeFiles/PicoPeripherals.dir/all] 错误 2
make: *** [Makefile:136：all] 错误 2

closed 2024-10-18 by arntsonl

mikepparks, 2024-10-18:
Yeah. This is an artifact of using pico-sdk 2.0. It's advisable to not use pico-sdk 2.0 against main until further notice.

chiyu521, 2024-10-20:
SDK 1.5.1 Build OK

zrainx, 2025-04-24:
I have the same building error with pico SDK 2.1.1.
finally,  I modified  source code and build ok.

lib/pico_pio_usb/src/usb_rx.pio.h  
...
+ #if 0
   static __always_inline void pio_sm_set_jmp_pin(PIO pio, uint sm, uint jmp_pin) {
   pio->sm[sm].execctrl =
      (pio->sm[sm].execctrl & ~PIO_SM0_EXECCTRL_JMP_PIN_BITS) |
      (jmp_pin << PIO_SM0_EXECCTRL_JMP_PIN_LSB);
 }
+ #endif

zrainx, 2025-04-24:
I found something at
https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/126