OpenStickCommunity/GP2040-CE #1206: Macro loaded incompletely after loading from a preset

cliu0507, 2024-11-01:
### Expected Behavior

The macro settings should be loaded correctly if user saved a preset to *.gp2040

### Current Behavior

Some instructions are lost after loading a preset saved before. My observation is that it can only load the first 4 instructions for some reasons. 

### Context

This issue doesn't occur in 0.7.8

TheTrainGoes, 2024-11-06:
Hi @cliu0507,

Was the macro made on a different version of the firmware?

cliu0507, 2024-11-07:
@TheTrainGoes Hi, yeah I double checked the version of firmware. It is actually  v0.7.9-8-h09 so i edited the title of this issue. Are I supposed to report issue on 0.7.9 still ? I meant the latest is 0.7.10

closed 2024-12-23 by cliu0507

TheTrainGoes, 2024-12-23:
Hi @cliu0507 ,

We only take issues on the latest version of the firmware.  If this issue is still in v0.7.10 please make a new issue.