OpenStickCommunity/GP2040-CE #1207: up and down revert

dongcidacia, 2024-11-04:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on firmware capabilities and usage.
> * Make sure that your idea has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).
> * Confirm the feature does not exist in the latest version of the firmware, found
  [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases).

### Problem

> Which problem is this feature request solving?
>
> Provide a clear and concise description of what the problem is. For example, "I'm always frustrated when..."

### Solution

> Describe the solution you would like to see
>
> Provide a clear and concise description of what you want to happen. Let us know about other solutions you've tried or
> researched.

### Do you plan to work on this feature yourself?

> Beyond a simple yes or no, we are also willing to help with partial solutions or to refine a proof of concept, if you
> are able to contribute.

TheTrainGoes, 2024-11-04:
Hi @dongcidacia ,

There is nothing in this issue.  Was it submitted in error?

TheTrainGoes, 2024-11-06:
No follow up on message

closed 2024-11-06 by TheTrainGoes