OpenStickCommunity/GP2040-CE #1213: ? defective GP16 on my Raspberry Pi Pico W board. How can I use remaining GPIO pins as select button

chikatana, 2024-11-06:
I'm new to the GP2040-CE project, thanks for all the hard works.

Long story short, I though my Raspberry Pi Pico W boards from Aliexpress maybe defective
and the GP16 pin is always registered as 1 even if nothing is connected to that pin. So the select key is always firing.

I've got some basic knowledge in Java but not the language which GP2040-CE is written in.
I'm thinking about editing the source code, replacing the function of GP16 with another unused GPIO pin, eg. GP 22,26,27 or 28 and then compile my own firmware.

Since I only know Java but nothing in the language which GP2040-CE is written in, I wound be very pleased if someone can tell me which file in the source code do I need to edit so that the GP16 pin can swapped into another pin.

Thanks!

TheTrainGoes, 2024-11-06:
Hi @chikatana ,

There is no need to re-compile the firmware for this.  

Simply boot into web-config, un-map GPIO16 from `S1` and choose another pin to act as `S1`.

```#define GPIO_PIN_16 GpioAction::BUTTON_PRESS_S1```

If you are having trouble getting into web-config due to `S1` being held then you can put the unit into BootSel mode and drop the force web-config .UF2 on which will make the unit reboot into web-config mode.  

you can find the force web-config .UF2 here:
https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.10/force_webconfig.uf2

closed 2024-11-06 by TheTrainGoes

chikatana, 2024-11-06:
Problem solved. Many thanks.