OpenStickCommunity/GP2040-CE #1223: EGRET II mini will not boot if GP2040-CE(EGRET2 mode) is connected to USB 2P side

tamanegitaro, 2024-11-25:
### * Confirm your issue exists on the latest version of the firmware, found
Yes. It happens on 0.7.10.
Also tried 0.7.6 and same issue occur.

### Expected Behavior
When GP2040-CE is connected to USB 2P side of the EGRET II mini console, it should be running as controller/joystick for player 2.

### Current Behavior
When GP2040-CE is connected to USB 2P side of the EGRET II mini console, EGRET II mini will not boot.
I have tried EGRET II mini without any SD card inserted.
Also tried EGRET II mini with Arcade Memories 2 SD card inserted.
Issue occurred on both cases.

Looks like when EGRET II recognizes GP2040-CE before system boot, system stalls somehow.
To get around this issue, I have inserted code in GP2040::setup() as below.
This code adds 3.7s delay before booting GP2040-CE when mode is EGRET II mini.

[image]

I am not sure if this is best solution, but confirmed that it works for both EGRET II mini with and without Arcade Memories 2 SD card.

### Context
* Firmware version (please get the full version listed in webconfig, if at all possible)
Firmware version is 0.7.10 latest version available on GP2040-CE website.
https://gp2040-ce.info/downloads/

* Board config name (also from webconfig, please)
Pico
Waveshare RP2040-Zero

* Name of device
Pico
RP2040 Zero

* Link to where you purchased the device
https://ja.aliexpress.com/i/1005007393402333.html

### Steps to Reproduce

### Please provide detailed steps for reproducing the issue.
1. Step 1
Configure GP2040-CE as EGRET II mini mode.

2. Step 2
Connect GP2040-CE device to EGRET II mini 2P USB port

3. Step 3
Boot EGRET II mini, system will not boot.

mikepparks, 2024-11-25:
This feels like it may be because the EGRET II Mini is looking for boot devices on 2P.  I know that a similar thing would happen on the Neo Geo Mini.  Unfortunately, this is not one that can really be tested without actual hardware, as it seems related to the device's function, and not the firmware's function as a controller.  None of the core team have this hardware to test against.

tamanegitaro, 2024-11-26:
> This feels like it may be because the EGRET II Mini is looking for boot devices on 2P. I know that a similar thing would happen on the Neo Geo Mini. Unfortunately, this is not one that can really be [...]

Hi Mike. I think I found cause of the issue. For official Egret II mini controller pad, USB descriptor was sent every 8ms even when the buttons are idle. GP2040-CE’s Egret 2 mini driver sends USB descriptor only when button is pressed or joystick is moved. Egret 2 mini was waiting for USB descriptor and not waiting for boot devices.

I fixed the code for Egret II mini driver so that USB descriptor will be sent every 8ms just like official Egret2 mini pad and looks like issue disappeared. I’ll test it this weekend too to make verification more reliable. After that, would you please let me report details of how I modified code so that you can make choice to accept it or reject it again?

tamanegitaro, 2024-11-29:
Hi Mike.

I have very sad news for you.
I modified the GP2040-CE Egret II mini driver so that USB descriptor will be sent every 8ms just like official Egret2 mini pad. Verified it on Egret mini2 20 times and it didn’t fix the issue. Issue still exists.

I can’t come up with any other idea to fix this issue so please close this thread for now.
One day if I find root cause and find how to solve it, let me submit another issue.

Thanks,
tamanegi_taro

closed 2024-11-29 by tamanegitaro