OpenStickCommunity/GP2040-CE #1224: Add docs on creating configuration pages for add-ons

Milk-Cool, 2024-11-26:
### Problem

Right now I am developing an add-on for taiko drum controls for a DIY taiko drum (basically just presses certain buttons on reaching certain thresholds on certain ADC pins) and there are no docs on creating a config page for a plugin. This makes it difficult for developers to create add-ons for contributing or for their own purposes.

### Solution

Just write docs for creating config pages for addons. That will make the lives of everyone working on GP-2040 a little easier.

### Do you plan to work on this feature yourself?

I honestly have no idea how all of this works, I was barely even able to write an addon by myself. So sorry, but I can't help with this one :(

Milk-Cool, 2024-11-26:
okay, i'm now rereading this and i'm sorry if this sounds a little rude, especially the solution section. my english's not that good tbh

TheTrainGoes, 2025-07-02:
Hi @Milk-Cool,

In order to have docs on how to create add-ons we need someone to make those docs.

closed 2026-11-05 by TheTrainGoes