OpenStickCommunity/GP2040-CE #1230: Directional inputs switch to analogue sticks in keyboard mode

willsy32, 2024-12-09:
I have function/hotkeys turned off and even deleted all hotkeys.  And also how is it possible to switch to analogue stick mode in keyboard mode?  I use this board for stepmania and at random if  I press a combination of two keys (I think up and shift, I really haven't figured out exactly what replicates it since it happens about 1/10 times I play).  Is there a way to just remove the analogue stick option as directional input when in keyboard mode?  I'll try this week to try to replicate what exact combination of keys I press, but otherwise I hope the issue I wrote makes sense.  I've had to turn off analogue mode about 10 times since I've used this firmware.

TheTrainGoes, 2024-12-10:
Please work on replicating the issue before posting like this.  

Also please try to be more clear when posting.

TheTrainGoes, 2024-12-10:
This will be assigned the big label if we determine it to be a bug and not user error.

TheTrainGoes, 2024-12-28:
Closing as we never heard back from OP.

closed 2024-12-28 by TheTrainGoes