OpenStickCommunity/GP2040-CE #1232: Gp2040 random inversions association Player 1 and Player 2 with 2 identical encoders pi pico.

Jimizan, 2024-12-13:
Pizero Gp2040 latest firmware , any model.
my are the Pi Pico 2020 originals bought on Amazon.
(fw. GP2040-CE_0.7.10_Pico.uf2)
when used for 2 players there are 2 identical models
even the deviceGUID is identical to the PC.
and even if they are no longer detached from the relative USB ports, RANDOM and often they are inverted on numerous player1 games with player2.
this happens with batocera, retroarch and others.
Confirmed with dozens of new users of the Gb2040, where in their bartop and PC.. they mount 2 encoders of these.
The hope where possible if through the firmware, it was possible to distinguish the deviceGUID between 2 or more identical Pi Pico and equipped with the same FW 2040.

TheTrainGoes, 2024-12-13:
Hi @Jimizan ,

This is incredible hard to follow.  

Please re-type the issue in clear and concise English using proper terminology.  

Also please state your source for `Confirmed with dozens of new users of the Gb2040`.

TheTrainGoes, 2024-12-13:
I think you are simply asking for a way to change the PID/VID/deviceGUID in the firmware so you can use two RP2040 devices running GP2040-CE at the same time.

This is not currently possible.  

You are best to run them in different modes for the time being like XINPUT on one and HID-USB on the other.

closed 2024-12-13 by TheTrainGoes

Jimizan, 2024-12-13:
yes, thanks that's exactly what I meant... sorry for the English..not much English.=)
as you surely know many use these encoders to play
but if there are two identical ones with the same GUID the computer sometimes confuses them as priority/usb and you find games with the player 2 commands on 1, and vice versa...practically inverted.
it doesn't always happen but often and randomly.

Using two identical original usb pads from the ps3 or xbox360 doesn't happen because they still have different GUIDs, precisely to avoid creating these "inverted overlaps"...

we hope it will be possible to add this option soon.
thanks for the support