OpenStickCommunity/GP2040-CE #1238: 1.3" I2C OLED LCD SH1106 draws past the display on the right side

exelion001, 2024-12-22:
### Expected Behavior
A .96" OLED using SSD1306 shows the complete buttons on the right.

### Current Behavior
The right buttons draw past the screen and are missing about 15% of the buttons.
This was used on 2 different pico boards with the same results.

Name of device
DIYUSER 1.3" IIC OLED Display Module White Blue Color Drive Chip SH1106 128X64 1.3 Inch LCD IIC I2C Communicate For Arduino
Link to where you purchased the device
https://www.aliexpress.com/item/1005007553189902.html

### Steps to Reproduce
1. Attach I2C display
2. Plug GP2040 into usb port

### Screenshots & Files
> Please include any relevant screenshots or files here, including photos of the front and back of the board.
[image]

mikepparks, 2024-12-23:
Which layout are you using?

exelion001, 2024-12-23:
I updated the steps to reproduce.
Button Layout (Left) - Stick
Button Layout (Right) - Fightboard

mikepparks, 2024-12-23:
Do the other layouts expand to the full width of the screen? I tested this one and it looks okay on all of my displays.

exelion001, 2024-12-23:
I can only notice with layouts that have more buttons.
ARCADE
[image]

BUTTONS ANGLED
[image]

CAPCOM
[image]

exelion001, 2024-12-23:
I tried a plash screen from [here](https://gp2040-ce.info/web-configurator/community-splash-screens) and it looks fine.

[image]

TheTrainGoes, 2024-12-23:
Tested this on a few of my displays as well and all look well.

@exelion001 with the above splash screen it is missing a the last column of pixels.  

You said above that you tested a pixel draw at (127,63), what happens at (128,64)?

exelion001, 2024-12-23:
I tried (128,64) but it doesn't show.  I think (128,64) is off the display because the first pixel is at (0,0).

TheTrainGoes, 2024-12-23:
@exelion001 OK just checking.  Yes, I have a number of those displays and they work as intended.

Do you have multiple of these displays exhibiting the same issue?

exelion001, 2024-12-23:
I only have the one display I got today.  You are right and it looks like I am missing 2 columns of pixels on the right in the splash screen.  I haven't seen many people use the 1.3" displays.

TheTrainGoes, 2024-12-23:
@exelion001 can you take a pic of the back side?

exelion001, 2024-12-23:
[image]
I ran some more tests in arduino using the following code:
  display.drawPixel(0, 0, SH110X_WHITE);
  display.drawPixel(127, 0, SH110X_WHITE);
  display.drawPixel(0, 63, SH110X_WHITE);
  display.drawPixel(127, 63, SH110X_WHITE);
  display.display();
  delay(5000);

  // draw a single pixel
  display.drawPixel(10, 10, SH110X_WHITE);
  display.setTextSize(1);
  display.setTextColor(SH110X_WHITE);
  for(uint8_t i=0;i<128;++i)
  {
    display.setCursor(0, 0);
    display.println(i);
    display.drawPixel(i, 10, SH110X_WHITE);
    display.display();
    delay(100);
    display.clearDisplay();
  }

The 4 pixels show in each corner for the first test and the second test draws a pixel across the screen correctly.

mikepparks, 2025-01-02:
I managed to find my collection of these exact units and I can reproduce this. Will see about getting a fix included for next version.

mikepparks, 2025-02-06:
This had been done in #1254

closed 2025-02-06 by mikepparks