OpenStickCommunity/GP2040-CE #1245: 30HZ turbo fire bugged?

psymage, 2024-12-27:
So I'm playing a shmup with a safe spot during a boss fight that requires the player to be in the top middle of the screen holding down 30HZ autofire for a little while in order to chew through the final phase of the fight. I have Turbo enabled set to 30HZ and mapped to one button but whenever I hold the button down, it seems like it goes down to a much slower rate after several seconds and sometimes, it seems to happen sooner. This much slower rate will occur for two to three seconds before resuming with 30HZ. This is in a section of the game where there isn't any frame rate problems either that could cause frames to get dropped. This is a little bit annoying and it doesn't appear in other footage of the game that I could find.

I even tried 29HZ and while it is slightly slower rate obviously, it does not have this issue as the rate of fire stays consistent even when holding it for several seconds at the top of the screen in the same area of the game. I wonder if the 30HZ setting is sometimes trying to do more than 30 presses per second for some reason, but that's my only theory and that could be wrong. 

I'm brand new to the GP2040CE firmware as I've only been playing on a fightstick with the RP2040 Advanced Breakout Board for about a month. I got the board off of FocusAttack. 

**Firmware:** 

RP2040 Advanced Breakout Board - USB Passthrough (GP2040-CE_0.7.10_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2)
Current: v0.7.10-2-g230983f
Latest: v0.7.10

tonytoon, 2025-01-05:
Which game and platform? It is possible the game is not running at exactly 60 frames per second which would make a naïve 30 Hz auto fire out of sync in certain edge cases.

psymage, 2025-01-05:
Gun Frontier in MAME, Stage 5 Boss

In that particular section of the stage, it doesn't have a frame rate dip. Here's Jaimers planning that particular section I'm referring to for further context:

https://youtu.be/Ii96536yzWo?si=FH_BnXGAZJpzckMA&t=1209

See his 30HZ turbo fire stays a constant stream until his shot makes contact with something. Mine would drop off to barely any shots for a few seconds for no apparent reason. 

Plus, the problem doesn't happen if it's set to 29HZ, which is very similar technically. If the game did have a significant frame rate dip, it would drop off at lower turbo settings as well. 

Oh, and I did test this after posting with another game, Metal Black, and had the same issue.

tonytoon, 2025-01-05:
Hmm, MAME reports the game as running at exactly 60hz. I'm going to poke at this one as an excuse to better learn the codebase and see if I can determine what's going on. If you use MAME's built in advanced autofire does it have similar issues?

psymage, 2025-01-05:
I actually don't use it. I use RetroArch's MAME Current core because I hate standalone MAME's UI so much.

arntsonl, 2025-02-17:
Verified we are working just fine. This did give me an opportunity to rewrite some of our add-ons, but our turbo is good!

https://github.com/user-attachments/assets/a0d93258-6d6c-4774-8b49-941f5708a216

closed 2025-02-17 by arntsonl

psymage, 2025-02-17:
No, the turbo has not been fixed. This is still a problem at 30HZ, just tested it in game again. This happens on not just one but two different PCBs running the latest GP2040CE firmware. It doesn't happen at 29HZ or lower, only 30.

bsstephan, 2025-02-18:
Given the test suite used above, I don't think we can go on in-game results alone, and would need a similar test suite to pinpoint an issue. If you can reproduce the issue in a clear and controlled environment, please let us know.

psymage, 2025-02-19:
> Given the test suite used above, I don't think we can go on in-game results alone, and would need a similar test suite to pinpoint an issue. If you can reproduce the issue in a clear and controlled environment, please let us know.

I'm going to make a video showing the same thing happen in multiple emulators with different shmups that can handle 30HZ autofire.

psymage, 2025-02-20:
I now have the video evidence proving the 30HZ turbo fire bug. So a shmup developer named Aktane made an input testing program in Pico 8 just to test this out and once again, the 30HZ autofire drops off significantly at times. I held the turbo fire down for about 40 seconds. 

https://github.com/user-attachments/assets/953e723e-169a-4ff0-b9c5-74351b6ce065

Meanwhile, 29HZ doesn't exhibit this same drop off. Sure, there are some smaller gaps in the shot because it's not 30HZ, but that is to be expected from 29HZ autofire running in a 60FPS program like this and is consistent with the behavior I've seen in-game when playing Gun Frontier with this turbo setting.

https://github.com/user-attachments/assets/3d9c54b8-1dfb-486d-881a-6e31c192178e

At this point, there's no reason this should stay closed. This is a real issue.

TheTrainGoes, 2025-02-20:
@psymage what controller / devices, if any, do you have that pass the above test?

psymage, 2025-02-20:
> [@psymage](https://github.com/psymage) what controller / devices, if any, do you have that pass the above test?

Besides my two GP2040-CE fightsticks, I do not have anything else that could test at 30HZ. I can give you the test program I'm using to try it out for yourself, and anyone else to welcome to try to recreate what happened to me. Two different versions because I don't know if you're on Windows or Linux. 

[default_windows.zip](https://github.com/user-attachments/files/18895301/default_windows.zip)

[default_linux.zip](https://github.com/user-attachments/files/18895304/default_linux.zip)

Now I do have a couple of other controllers that can do 15HZ but that wouldn't be a very good comparison.

arntsonl, 2025-02-20:
Please find a controller that works, otherwise we have nothing to test / compare against. As far as I know, 8bitdo, Mayflash, MadCatz, and all of the others will have the exact same issue you're describing because the hardware does not sync to the software. If there is a controller that can work at 30hz, let us know.

psymage, 2025-02-21:
> Please find a controller that works, otherwise we have nothing to test / compare against. As far as I know, 8bitdo, Mayflash, MadCatz, and all of the others will have the exact same issue you're [...]

That's a much taller task than you realize. 8bitdo runs at 15HZ, and I know this because my SN30 Pro Plus was also at 15 for turbo and so is the M30 which I know from Aktane telling me. My Qanba Crystal only does 15HZ. Even my old Brook PS4 PCB would only go up to 25HZ max. Mayflash sticks only go up to 25HZ. Same with the older Madcatz sticks IIRC.

The only other PCB I could possibly think of that might go to 30HZ would be Bit Bang's TKO Board, and those are sold out right now.

In fact, part of the reason why I bought these PCBs in the first place was to have 30HZ autofire that doesn't depend on the emulator itself to set it up. Plus, it does not appear that rates lower than 30 are an issue.