OpenStickCommunity/GP2040-CE #1258: arrow keys in keyboard mode

DRAKE9800, 2025-01-06:
### Expected Behavior

Keyboard config arrow key pins should work

### Current Behavior

Controllers arrow key function not working **after .NET framework update**, If configured as XBOX controller the pins trigger normally.
Tested on two windows machines, same issue.

Version
Pico-W (GP2040-CE_0.7.10_PicoW.uf2)
Current: v0.7.10-2-g230983f
Latest: v0.7.10
Memory (KB)
Flash: 967.02 / 2048 (47.22%)
Heap: 20.94 / 137.17 (15.26%)
Static Allocations: 126.83
Board Flash: 2048

ahpinkor, 2025-01-19:
I had a rather similar experience in keyboard mode, when in-game, I suddenly lost all 4 direction inputs, but other buttons were working fine.  I suspected I might have accidentally triggered one of the hotkeys that changed the dpad mode (S1 + S2+ one-direction).  This dpad mode is preserved across power-cycles. In the end, I just reconfigured, and disabled hotkeys.

TheTrainGoes, 2025-08-12:
We have not been able to verify this on our end.

Theres a chance the input mode was changed from DPad (arrow keys / WASD) to Left / Right analog mode which would result in them looking like they were not working.

Past that DPad directions work as expected in Windows on the most recent updates (Windows 11 with all installed updated + .NET)

closed 2025-08-12 by TheTrainGoes