OpenStickCommunity/GP2040-CE #1265: USB Host  crash bug

fangxiang-gan, 2025-01-08:
I am using the gp2040-ce_0.7.10_picofatingboard.uf2 firmware, and I have repeatedly verified this bug. Specifically, when setting up the USB Host, if Pin Order is D+/D -, it can be saved normally. If Pin Order is D -/D+, the chip will crash and cannot be saved. Please fix it, thank you.
[image]
[image]

fangxiang-gan, 2025-01-08:
make corrections：I am using the
GP2040-CE_0.7.10_PicoFightingBoard.uf2

fangxiang-gan, 2025-01-08:
Yes, if configured like in picture two, the entire chip will freeze and not respond. It is necessary to unplug and power off again. Re enter the webpage configuration backend to check, the previous changes are invalid.
After flashing the firmware with a brand new one, except for the item in picture 2, the problem can be reproduced without any other changes, 100%.

fangxiang-gan, 2025-01-08:
Additional note: As I need to use pins 18 and 19, I have set the Player LED Type to the off state. As shown in the picture
[image]

TheTrainGoes, 2025-01-08:
Do you have another device to test this on?  I would have to check but I believe that the PFB has resistors on those GPIO unless you are using a clone or version of it without them.

fangxiang-gan, 2025-01-08:
My newly purchased Pico will arrive tomorrow, and I will test it again and provide you with specific feedback on its performance. Thank you for your reply.

fangxiang-gan, 2025-01-09:
Confirmed issue, as long as it is set as follows, it will fail to save and fake death.
[image]

The brand new genuine chips I use are PICO and PICO H.
[image]
[image]

TheTrainGoes, 2025-01-09:
Hey @fangxiang-gan ,

On the base Pico config GPIO18 and GPIO19 are set for use with buttons:

```#define GPIO_PIN_18 GpioAction::BUTTON_PRESS_L3     // L3     | LS     | LS      | L3       | 11     | LS     |
#define GPIO_PIN_19 GpioAction::BUTTON_PRESS_R3 ```

Are you unmapping those prior to this change?

fangxiang-gan, 2025-01-09:
Hey@TheTrainGoes
GP2040-CE_0.7.10_PicoFightingBoard.uf2
#define GPIO_PIN_18 GpioAction::PLED #3 Pin
#define GPIO_PIN_19 GpioAction::PLED #4 Pin

I have set the Player LED Type to the off state. As shown in the picture
[image]