OpenStickCommunity/GP2040-CE #1269: bluetooth or any wireless connection support?

MasterAlive, 2025-01-14:
I just came across this repository and made an R-pico X-input controller for myself.
This project is massive and has some settings that blow my mind.
The only thing missing is wireless connection support.
I was wondering if is it possible to connect a Bluetooth module such as hc-o5 or not.
Maybe I get to work with it in the future but I just want to post it here so that someone gets the idea and if it's easy to do for themselves, add this future sooner.

TheTrainGoes, 2025-01-14:
Hi @MasterAlive,

I'm not aware of anyone actively working on BT or wireless support so you are welcome to play around and let us know what you come up with.

ELMERtem, 2025-05-10:
well there is a workarond for ble support 
            |
            |  
           V
     [     click to see what i'm refuring to](https://www.youtube.com/watch?v=dI-08resits&t=13s)

ELMERtem, 2025-05-10:
code for esp32 S2/3
#include <BleGamepad.h>
#include <BLEDevice.h>

BleGamepad gamepad;
HardwareSerial &uart = Serial1;

void setup() {
  uart.begin(115200, SERIAL_8N1, 9, 10); // RX=9, TX=10
  gamepad.begin();
}

void loop() {
  if (uart.available()) {
    String line = uart.readStringUntil('\n');  // Read packet from Pico

    // Macro button handling
    if (line.startsWith("BTN:")) {
      String cmd = line.substring(4);
      if (cmd == "START") {
        gamepad.press(BUTTON_START);  // Simulate press on Start button
        delay(100);  // Hold for a short time
        gamepad.release(BUTTON_START);
      } else if (cmd == "MACRO_A") {
        gamepad.press(BUTTON_A);  // Simulate press on Macro A (or other buttons)
        delay(100);
        gamepad.release(BUTTON_A);
      }
    } else {
      // Read touchpad x,y,mode data
      int x, y, mode;
      sscanf(line.c_str(), "%d,%d,%d", &x, &y, &mode);

      if (mode == 0) {  // Mouse mode
        gamepad.mouseMove(x, y);  // Use x, y for mouse movement
      } else {  // Joystick mode
        gamepad.setRightThumb(x - 128, y - 128);  // Map x, y to joystick
      }
    }
  }
}