OpenStickCommunity/GP2040-CE #1281: PS4 Forced Mode using HOTKEYS causing unexpected behavior for D-PAD inputs. Using WEBMODE to set PS4 mode, D-PAD Operates as expected

JasensCustoms, 2025-02-01:
### Expected Behavior

> Pressing PUNCH 2 on plugin places Pico Fighting Board into PS4 mode.  Directional inputs in Windows JOY.CPL work properly.

### Current Behavior

> Pressing "PUNCH 2" to force the Pico Fighting Board into PS4 mode, D-PAD directions do not operate properly. Using Windows JOY.CPL shows that UP+LEFT is pressed. Pressing any other directional inputs causes the indicated direction to change, but not correctly. All other buttons work correctly.

> When forcing the Pico Fighting Board into PS4 mode via web config, D-PAD directions work as expected in Windows JOY.CPL.

> Pressing "KICK 2" to force the Pico Fighting Board into XBOX (XINPUT) mode, D-PAD directions work properly. When using Windows JOY.CPL all directions work properly. All other forced modes for the PICO FIGHTING BOARD works correctly.

### Context

> * Firmware version: Fresh download of the Pico Fighting Board V0.7.10
> * Hardware: Integrated Pico Fighting Board (All Versions)
> * Purchased From: http://www.jasenscustoms.com
> * Windows 11
> * Chrome

### Steps to Reproduce

> Using button HOTKEY method to FORCE MODES
>
> 1. Plug the Fighting Board in while holding P2
> 2. Observe Fighting Board enumerates in JOY.CPL
> 3. Click the Properties button, observe the input lights on JOY.CPL and current status without anything pressed
> 4. Press buttons and verify button inputs work

> When using WEBCONFIG to place board in PS4 Mode.
>
>1. Plug the Fighting Board in while holding START
>2. Navigate to Web Config
>3. Set Joystick to PS4 mode.
>4. Click Save.
>5. Reboot to Controller Mode.
>6. Observe the fighting board enumerates in JOY.CPL
>7. Click the Properties button, observe the input lights on JOY.CPL and current status without anything pressed
>8. Press buttons and verify button inputs work (they do).

Using WEBCONFIG MODE, NO BUTTONS PRESSED:
[image]

Using FORCED MODE, NO BUTTONS PRESSED:
[image]

TheTrainGoes, 2025-02-02:
Several of us tested this and were not able to replicate given the information above.  

It is not clear if you used our `GP2040-CE_0.7.10_PicoFightingBoard.uf2` firmware to test this or your own.  

Were any additional changes made to the settings or are you using the defaults of the PicoFightingBoard?

My test:
- Fresh Pico
- Nuke
- Fresh download of PicoFightingBoard put on
- Unplug
- Replug with P2 held down

I also tested setting as the various modes of PS4/5 and could also not replicate.

Possible issue with the Integrated Pico Fighting Board or custom firmware (if being used).

JasensCustoms, 2025-02-02:
1. This is a fresh, new PCB.
2. There is no custom or modified firmware for the IPFB, it uses stock GP2040-CE firmware downloaded from the GitHub. Specifically, the PicoFightingBoard V0.7.10 UF2 as stated above. To ensure there were no conflicts, no changes were made in webconfig after the flash except when swapping modes to replicate behavior.
3. The same steps above were followed and the issue persisted.
4. This only occurs in the PS4 Forced Mode using P2 being held down; all other forced modes operate correctly.
5. The IPFB follows the RP2040 design guide exactly as written by the Pi Foundation and operates as intended in all other forced modes or when pushed to PS4 using the webconfig.

TheTrainGoes, 2025-02-02:
I have tested this on the following devices with the following firmwares and have not been able to replicate:

1 - Haute42|COSMOX S16 -> GP2040-CE_0.7.10_Haute42COSMOX.uf2
2 - RP2040 Advanced Breakout Board v5.6 (Version 2) -> GP2040-CE_0.7.10_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2
3 - Raspberry Pi Pico -> GP2040-CE_0.7.10_Pico.uf2
4 - Raspberry Pi Pico -> GP2040-CE_0.7.10_PicoFightingBoard.uf2

I also loaded GP2040-CE_0.7.10_PicoFightingBoard.uf2 onto the IPFB 'November 2022' board I have and was not able to replicate.  

Of note here is that when I attempted to put GP2040-CE_0.7.10_IntegratedPicoFightingBoard.uf2 from https://github.com/JasensCustoms/GP2040-IPFB/releases/tag/v0.7.10 onto the IPFB 'November 2022' or a stand alone Raspberry Pi Pico it caused the unit to lock up and would not show up in either Windows JOY.CPL or the online hardware tester (https://hardwaretester.com/gamepad).

---

Re: #2 - It is my understanding that the IPFB runs on a custom configuration of GP2040-CE as available from https://github.com/JasensCustoms/GP2040-IPFB/releases/tag/v0.7.10 under the name GP2040-CE_0.7.10_IntegratedPicoFightingBoard.uf2.  Outside of this I was able to load our hosted GP2040-CE_0.7.10_PicoFightingBoard.uf2 onto the device and could not replicate the issue.

Re: #5 - The IPFB is a closed source device and not supported by the project.

---

If you are able to replicate this on a device that we support like the Raspberry Pi Pico running our hosted GP2040-CE_0.7.10_Pico.uf2 please let us know and we will investigate further.

TheTrainGoes, 2025-02-02:
Shortly after my last comment I can see that the IPFB was open sourced via https://github.com/JasensCustoms/Integrated-Pico-Fighting-Board?tab=readme-ov-file.

I was able to download both the board file and the schematics to EasyEDA and take a look. 

The converted import into EasyEDA produced a number of DRC errors but this is common so they were omitted during my look.

There only thing that stood out to me on first pass was the crystal oscillator (C409301).

The provided schematics (https://github.com/JasensCustoms/Integrated-Pico-Fighting-Board/blob/main/eagle-cad-files/integrated-pico-fighting-board-gp2040-layout-Version3.0.sch) Only reference XIN being connected for the oscillator where the RP2040 Hardware Design document (https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf) shows XIN and XOUT connected to the crystal oscillator.

I am not familiar with the crystal oscillator used in this design (C409301), only C9002 which I use on my own designs.  

There was a period of time a few years ago where some crystal oscillator related issues were observed at startup which could causing unexpected behaviour.  We addressed this by adding a small delay on startup.  I am unsure if this specific crystal oscillator may also be causing issues.  

---

Outside of this it would be important to know if this can be replicated on any boards other than the IPFB.

JasensCustoms, 2025-02-02:
This design uses a CMOS oscillator, not a crystal. The use case was validated by the Director of Engineering at Abracon who provides the actual CMOS oscillator used in my builds (the LCSC part was the closest available for the BOM to ensure there was no questions about the design, but it also works the same way). This was a change based on the slow startup of some crystal oscillators you noted causing issues. I consulted them to ensure I used their device properly since it was a significant change. As such CMOS oscillators do not require a connection to to the XOUT as they do not have an OUT pin.

From the design doc: 

"Providing an external frequency source can be done in one of two ways: either by providing a clock source with a CMOS
output (square wave of IOVDD voltage) into the XIN pin, or by using a 12MHz crystal connected between XIN and XOUT.
Using a crystal is the preferred option here, as they are both relatively cheap and very accurate."

I don't use other boards using the GP2040-CE firmware, so I can't comment if this can or can't be replicated.

TheTrainGoes, 2025-02-02:
Does this happen consistently on the IPFB or sporadically?

If consistently can you please do a UF2 flash dump of your IPFB that exhibits the issue and I can test on other RP2040 based devices.  

Are there any other factors which may be affecting the operation of the IPFB such as connected accessories / add-ons / etc (dongles, screens, LEDs, etc) or are you able to replicate this issue with an IPFB connected on its own to the computer?

mikepparks, 2025-02-12:
@JasensCustoms I'm going to take a moment to address an issue that came from this, well, issue.

We get a large number of support requests for "unexpected behavior", so we are trying to do our best to troubleshoot through what we can and can't see. The fact that your IPFB was a closed board prior to us mentioning that fact made it a little difficult to diagnose whatever issues that you were seeing.  From the behavior you were describing, and the lack of potentially important information that would help troubleshoot, our ability to troubleshoot this is a bit limited.  However, nowhere in this discussion was it ever suggested that you needed to "open your board" before we could troubleshoot.  We provided a number of scenarios outside the scope of what you were testing on to hope that you could potentially reproduce the issue on something that we _can_ see and hopefully better understand what the issue is.

Whether you open or close your board is 100% up to you.  We provided a path to hopefully better understand the issue you were having as none of us were able to reproduce the issue with the hardware that we all have on hand.  We cannot be expected to have every board and variant in existence, as much as some of us try to do in order to better support the community and support local makers.  So I do take great issue with the fact that on a number of your product pages that you call out our project specifically because we didn't approach your issue with white glove service.  It is unfair that you feel that you feel we're aiding in a "race for the bottom" because makers that you personally take issue with are making products that we get a SIGNIFICANTLY LARGE number of support requests for on a daily basis.

We are but a small group of people who do this in our spare time with no expectation of any reciprocation other than people enjoy the firmware and related works that we give freely to the community.  We do this because we enjoy making cool stuff.  We do this because we are proud of the knowledge and skills we pick up along the way.

The fact that you make comments in your product pages, of which you are making claims that we are not responsive to your concerns, about products we had no knowledge of throughout the course of this bug report, is quite unfair and a slap in the face to everyone who contributes to this project.  We try to support our firmware to the best of our abilities regardless of the product using it.  However, to pick up your ball, leave the Discord, post inflammatory commentary on your for-profit products, and go home, all because we asked you to provide us with more information that we could better troubleshoot your issue?  Honestly, that feels a little disrespectful and rude.  Reacting by opening your board, cutting off sales and production of that board, and complaining about it?  That's pretty over-reactionary.

We want to help you determine the cause of this issue.  However, you need to be a little more open with us when trying to present all of the details, or at least as much as you can be without "revealing all of your cards."  We get that you want to make sure that you can bring the best product forward that you can.  Same here.  Your one-sided commentary however?  Unnecessary.

mikepparks, 2025-02-12:
We welcome that information once it comes.  Naturally, trying to reduce any bit of latency, perceived or otherwise, is a high priority for us, so we want to make sure that it's something we're looking at.

I will say that I still highly disagree with your commentary that our project is directly hurting local makers.  Local makers have always been undercut by those with a means of production greater than our own.  That also assumes that we're not local makers ourselves. 
 That's just the nature of the things, and has always been that way since well before our project existed.  **To pin that blame directly on us and the greater GP2040-CE project by saying that we're celebrating the flood of low-cost devices is wholly misguided, and disrespectful to the hard work we have put towards this project.**

Clones and derivatives have always existed.  The fact that they can use open source software to be able to lower their costs of going to market has always been an issue, no matter what field you look at.  While it's unfortunate that local makers are losing profits due to the factories that can just crank out the Ali/Temu junkware we see everywhere, at the end of the day, having open solutions to common problems is better than only having one option and paying their ludicrous rising prices and dealing with vendor lock-in.  We all benefit from the lower costs of free, open software with a dedicated community behind it.  It means we all can focus our resources into the harder and more costly aspects of improving our products.

TheTrainGoes, 2025-02-17:
We have locked this issue and will be closing it due to being unable to reproduce through our own testing.  We do not believe there are any issues with the firmware that would be causing this.