OpenStickCommunity/GP2040-CE #1288: Cannot Boot Haute42 | Firmware not Installing

haider0072, 2025-02-23:
### Expected Behavior

> When I plug in my haute42 it should boot up like it always do.

### Current Behavior

When I plug in my haute42, it does not boot up, else it shows `RPI_RP2` drive in file explorer and I can't boot my board.

### Context

- My last version was `GP2040-CE_0.7.10_Haute42COSMOX.uf2`
Which was tested and working fine, I updated it a night before the tournament I had to go.

- For the tournament ps5 were used so I did not had a magic stick for ps5 compatibilty so a friend with a haute snackbox suggested to connect his xbox controller i think i dont remeber so that the 8 minute play time issue could be resolved.
- After plugging it in , the haute did not start working and boot up. I thought it was a port issue but I tested on my PC it detects it and only showns `RPI_RP2` in the file explorer. 

- I tried flashing it and installing the firmware.
- On flashing it, my haute42 shows the green blink on its led light which i think indicates it's being flashed.
- But on copying the firmware file, it does copy successfully but does not seem to work and boot my haute.
- It feels the firmware is not being installed.

### Screenshots & Files

[image]

[image]

https://github.com/user-attachments/assets/ca03a54a-527d-4668-a60d-e0e61720a694

haider0072, 2025-02-23:
[image]

TheTrainGoes, 2025-02-23:
@haider0072 ,

As I let you know in the Haute42|COSMOX Discord this likely has nothing to do with the firmware.  

If you have followed the recommended steps of putting the Nuke file on and then attempting to put the firmware on and it's still doing this it is a hardware based issue.  

I would connect with the seller of the device to look at replacement option.

closed 2025-02-23 by TheTrainGoes