OpenStickCommunity/GP2040-CE #1294: SPI Add-On (ADS1256) does not stay enabled upon refresh or reboot

tommybee456, 2025-03-11:
### Expected Behavior

When turning on the ADS1256 Add-On in the web config, it should stay enabled after a reboot or a refresh.

### Current Behavior

ADS1256 Add-On will disable itself upon refreshing or rebooting the page

### Context

I tested this on both a Pico(clone) and a waveshare rp2040-zero. I thought it may have something to do with how the data is stored but I can't seem to make sense of it. I tried recompiling the firmware to force the add-on to be enabled by default in the BoardConfig.h file:

>#define SPI_ANALOG1256_ENABLED 1
>#define SPI0_ENABLED 1
>whatever pins

but this makes the pico no longer boot. Maybe something is wrong with the SPI implementation? I originally though it had to do with the config_utils but that seems fine, so then I thought it had something to do with the WWW stuff, but In the browser console it shows the data correctly when hitting save so I am not sure what is going on.

<sub><sup>i know this Add-On is experimental but i am hoping i can at least enable it lol </sup></sub>

### Steps to Reproduce

1. Go into Web Config
2. Make sure SPI is enabled
3. Enable ADS1256 Add-On and Save
4. Refresh page or Reboot
5. Add one will have now disabled itself

### Screenshots & Files

[image]

[image]

[image]

TheTrainGoes, 2025-03-11:
I can replicate the lockup on my Haute42|COSMOX S13 by enabling SPI and the following pins.  

We will take a look when we can.

[image]

tommybee456, 2025-03-11:
@TheTrainGoes it looks like you have i2c0 still enabled with pins 0 and 1 which conflict with your spi0. Does it still happen with i2c0 disabled? Also were you able to reproduce the issue with the addon not staying enabled?

Anyway I appreciate the response.

mikepparks, 2025-03-11:
Let's try this. Make sure I2C0's SDA and SCL pins are set to Unset and turn off the peripheral. Then try SPI0. Does it crash?

tommybee456, 2025-03-11:
[image]
For the sake of it ill try different pins too

[image]
and then ill refresh the page

[image]
still disables itself

If I reboot it still works(enumerates as an xinput controller) but SPI is doesn't seem to be running. I don't see anything on SCK or RX or TX with my oscilloscope. CS is low, which it should be, however normally I would expect it to be high at boot, which it is not. SCK is high(low at boot), RX, and TX are low.

TheTrainGoes, 2025-03-11:
I totally botched the testing of this haha.

tommybee456, 2025-03-11:
The only time I got the Pico to crash was when I forced SPI to be enabled in BoardConfig.h and recompiled the firmware. I also had removed the #define for i2c and the sda and scl pins. I may have done something wrong there that also caused the crash however

mikepparks, 2025-03-12:
Please give the artifact [here](https://github.com/OpenStickCommunity/GP2040-CE/actions/runs/13803309122) a shot. This fixes an issue that was causing the addon to not be enabled/disabled and have its settings defined on save.

tommybee456, 2025-03-12:
@mikepparks That seems to be working! I only tested on my Pico so far but it both stays enabled and I can see the SPI pins doing stuff with my oscilloscope. Thank you very much!

You are a gentleman and a scholar :)

Should I close the issue or should I leave that to yall?

mikepparks, 2025-03-12:
I'll take care of it.

Turns out the problem was that the code that enabled the saving of config options had been broken for a while now, but not many of us use that particular addon, so it was a good find. Thanks for pointing it out.

closed 2025-03-12 by mikepparks