OpenStickCommunity/GP2040-CE #1301: Pin Viewer always show GPIO 26  &  Cannot Save Pin Mapping Profile

JJcmdAAAclzo, 2025-03-15:
### Context

> * Firmware: GP2040-CE_0.7.10_Pico.uf2 (v0.7.10-2-g230983f)
> * Board config name: Profile 1 (Default Name)
 
> * Device: Custom Leverless
> * Device Pin Wiring: https://www.helloimg.com/i/2025/03/15/67d58e5739b5a.png
> * Board Wiring: https://gp2040-ce.info/controller-build/wiring/#raspberry-pi-pico

> * Operating System: Windows 10
> * Browser: Microsoft Edge

### Current Behavior

> When using the Pin Viewer function at [Configurator](http://192.168.7.1/pin-mapping), Pressing the button always immediately show "Pressed Pin 26". Pressing "Map buttons with 🎮" will make windows quickly flash through, until Pin 26 is mapped with [Function].
> It is also impossible to create new Profile; Every time the device reboot or even switch page in browser configurator, any newly added profile will disappear, with only the default Profile 1 remain.

### Steps to Reproduce

---Pin Pressed Return---
> 1. Pressing the Start Button(Pin 17) and connect the device to the computer through USB-C to A Cable.
> 2. Open the [Pin Mapping Page](http://192.168.7.1/pin-mapping) in the browser
> 3. Press "Pin Viewer" button in the browser.
> 4. Pressed Pin Value Return: "Pressed Pin: 26".

---Saving Mapping Profile---
> 1. Open the [Pin Mapping Page](http://192.168.7.1/pin-mapping) in the browser
> 2. Add Profile in the website. New Profile will contain the same configuration with the default Profile 1. Make small adjustment--or not--and press "Save" button
> 3. Press "Setting" at the top of Configurator page, or reboot the device.
> 4. Re-enter the Configurator Website. The Profile should be gone.

JJcmdAAAclzo, 2025-03-15:
FollowUp:

- Re-Installing the firmwire does not resolve the issue.
- The device contain 2 USB-C port and 1 USB-A port(for PS5 Adpator Connection). Connecting the device to computer through USB-A port does not provide any response.

TheTrainGoes, 2025-03-15:
Hi @JJcmdAAAclzo ,

It is a little unclear what the specific issue is on this. 

Are you saying that no matter what you press Pin26 comes up?

JJcmdAAAclzo, 2025-03-16:
Sorry for inconvenience.

What I mean is that everytime I press the Pin Viewer Button in the website Configurator, Rather than waiting my input, it will immediately return "Pressed Pin 26".

Footage can be seem here: https://cdn.discordapp.com/attachments/994043723174387722/1350665692990410805/PinMapping.mp4?ex=67d790f3&is=67d63f73&hm=80f62b01695cdfe54fd40691dee84227ad2c9472698aad6cddd2cbd3411934fc&

JJcmdAAAclzo, 2025-03-16:
Footage of Pin-Mapping profile gone after created:  https://cdn.discordapp.com/attachments/994043723174387722/1350667771524022302/Profile_Lost.mp4?ex=67d792e2&is=67d64162&hm=c3568b4321d9e5811a2bc02b14f3faf0b6d21043cc32ddeec6217234003930bb&

TheTrainGoes, 2025-03-16:
This sounds like a combination of possible wiring issues and issues browser side.  

Can you post a pic of your internal wiring and also try the web-config in another browser or on another computer.

JJcmdAAAclzo, 2025-03-16:
Update:
Somehow I was able to create a new profile after switching to chrome, through mapping Pin 26 in Configurator, it is possible to create and save a new profile. 
[[Footage Here]](https://cdn.discordapp.com/attachments/994043723174387722/1350702370090385448/ChromeTest1.mp4?ex=67d7b31b&is=67d6619b&hm=ff5a010aa3af37de600ddf3dce89ace7025de7f293a67ef1e5d28c7ea5740c08&)

However, this method is very inconsistent and often result in failure to add or change the profile.
[Failed Attempt 1](https://cdn.discordapp.com/attachments/994043723174387722/1350717829267062784/Failed_Attempt2.mp4?ex=67d7c181&is=67d67001&hm=9f2a36872e042e0bed53412897757d18ed378a84a2b86247c406ae112cf97618&)
[Failed Attempt 2](https://cdn.discordapp.com/attachments/994043723174387722/1350717830055858257/Failed_Attempt.mp4?ex=67d7c181&is=67d67001&hm=256c16d04fe969eb47f9334ca3af2188430decae99a169b5dbf92b28250c8ba5&)

What is the possibility of the cause being hardware issue in my device? 
I will try to dissemble my leverless later today to check the wiring.

JJcmdAAAclzo, 2025-03-16:
[image]
[image]

Will these be Helpful?

TheTrainGoes, 2025-03-16:
@JJcmdAAAclzo these are not overly helpful as I don't know what your wiring is on everything.  

Generally when this happens there is a pin conflict that is preventing the save or there is an addon that is assigned to that same pin.

Please try and replicate on a raw Pico not attached to the board you are using.

TheTrainGoes, 2025-03-16:
Just noticed this is also a clone Pico.  Please try and replicate on a non-clone Pico

JJcmdAAAclzo, 2025-03-16:
[<Wiring Images>](https://imgur.com/a/Jknnxs2)

My apologies for causing any inconvience.
I actually purchase this device from a seller. Is it possible that my problems are due to the...sub-par production of clone-chip?

TheTrainGoes, 2025-03-17:
Hi @JJcmdAAAclzo ,

There are a number of things that can cause this issue.  

I have re-tested v0.7.10 on a stand-alone Raspberry Pi Pico and cannot replicate the issue you have.

The things that could be causing this are:
1 - Poor solder joint on the board that is causing a bridge somewhere
2 - Poor construction of the clone Raspberry Pi Pico clone
3 - Conflict in the config from the custom setup on that board (if it uses one)

These are just some ideas. 

We are going to close that as we cannot replicate on our end and I would ask that you work with the seller for support moving forward.

closed 2025-03-17 by TheTrainGoes