OpenStickCommunity/GP2040-CE #1379: Is it possible to use GP2040-CE device in combination with other?

pj-r, 2025-05-09:
I have been playing with idea to add linear hand brake to g29 steering wheel, pedals and shifter setup. Is it possible to use g29 and GP2040-CE side-by-side way the GP2040-CE would provide one extra analog input mapped L2, for example?

mikepparks, 2025-05-09:
Nothing officially planned at the moment, but definitely stay tuned.

pj-r, 2025-05-10:
Perhaps I could try to help with this. I am not a native in c++ but done some stuff with arduinos.
Some one needs to push me a bit to right direction. Is the "Gamepad USB Host" add-on the best way to begin with this?

There is already some implementation existing for older logitech steering wheels to be emulated as g29.
https://github.com/jfedor2/wheel-adapter

pj-r, 2025-05-10:
Oh, I see, you have already done some base work for g29 👍 

https://github.com/OpenStickCommunity/GP2040-CE/pull/1371/files#diff-957d924b19f605d9570a3cb9e64573bdcc9b437d6f65d062dd486fb65c8ab05dR28

mikepparks, 2025-08-16:
It's worth mentioning that it's not exactly working _with_ g29, but simulating some parts of the g29.

TheTrainGoes, 2026-11-05:
Closing as not planned

closed 2026-11-05 by TheTrainGoes