OpenStickCommunity/GP2040-CE #1380: Keep Pin Viewer open to test mulitple inputs quickly

eliasreid, 2025-05-10:
### Problem
[Pin Viewer](
https://gp2040-ce.info/web-configurator/menu-pages/pin-mapping?_highlight=pin&_highlight=viewer#pin-viewer-video_game) is great but can be cumbersome to use when you want to check the pins of multiple buttons, as you have to re-open the dialog with the mouse or keyboard for each button you want to test.

### Solution

My suggestion is to add a mode for this dialog that keeps it open and shows the inputs until the user closes it explicitly. The dialog could show a similar `Pressed pin: <number>` message as is done currently in the pin-mapping page after the dialog closes. 

This mode could be switched on in a couple different ways:
1. add a checkbox to pin-mapping page beside the "GPIO Pin Viewer" button, with label like "keep open"
2. add a checkbox to the pin viewer dialog. Once issue with this approach is that the dialog currently closes on its own by default if no buttons are pressed. This could lead to a frustrating experience of the dialog closing as the user tries to click "keep open". 

### Do you plan to work on this feature yourself?

Yes, I would like to if others agree it would be useful

TheTrainGoes, 2025-05-10:
Hi @eliasreid ,

This is already part of the firmware if you have an OLED connected to the deivce.  When in web-config mode press `K3` and you will go into a live view of the buttons and pins being pressed.

eliasreid, 2025-05-10:
Ah, I don't have an OLED screen on my controller unfortunately (yet 🙂).

Paxtez, 2025-05-21:
To add on to this, also making it show what button the pin in mapped to (with the selected display type) would be nice.

TheTrainGoes, 2025-05-21:
@Paxtez that is also already in the firmware if you have a connected display.