OpenStickCommunity/GP2040-CE #1390: Analog Multiplier

joppy-at-git, 2025-05-26:
### Problem and Solution

I would like if you can use one analog channel to multiply the output of a selected number of other channels. E.g. you can enable this setting, and with a analog trigger and a joystick, when you pull the trigger to 100% the output of the output of the joystick get reduced by 50%. If you pull the trigger 50% the joystick output get reduced by 25% The trigger would essentially only be visible for the MCU, and used to decide the output level of the joystick, and it doesn't exist for the computer. In this setting you should be able to set the range of it, e.g. if you want 100%-1% trigger pull to result in 100% reduction to 1% reduction, or 50% reduction to 1% reduction. You could also inverse it, making the output increase when you pull the trigger. If you can make none linear curves it would be even cooler. 

This feature would be useful as you can create something similar to a "gas pedal" in games which lacks one, which can make strafing in a wide game more similar to WASD, since you can have 100% speed even with a small movement as long as you have your trigger fully pulled (or not pulled, if you inverse it to act as a break). Or you can have a aim joystick with a wider range of output while still not having a physically larger joystick.

### Do you plan to work on this feature yourself?

I know too little to contribute.

TheTrainGoes, 2025-05-29:
Whats the use case for this feature @joppy-at-git ?

TheTrainGoes, 2025-05-29:
@joppy-at-git im looking for some specific examples in games.

closed 2025-06-12 by TheTrainGoes