OpenStickCommunity/GP2040-CE #1408: The guitar hero display does not match the regular button layout

mushinsha, 2025-06-21:
I am thinking of creating another branch of the GP2040 firmware just to change it myself but I think you all should know. 

This line here: https://github.com/OpenStickCommunity/GP2040-CE/blob/main/headers/buttonlayouts.h#L441

should be  {GP_ELEMENT_BTN_BUTTON, {  5, 26,  14, 36, 1, 1,    GAMEPAD_MASK_B1, GP_SHAPE_SQUARE, 0, 0, 0}},\

The first button on a guitar hero guitar is always B1 or 'A' or 'Cross' or 'Accept' but with the current configuration the first button is B2 or 'B' or 'Circle' or 'Deny/Back' in the menus.

Overall the lines from 441 to 445 should look like this instead:

{GP_ELEMENT_BTN_BUTTON, {  5, 26,  14, 36, 1, 1,    GAMEPAD_MASK_B1, GP_SHAPE_SQUARE, 0, 0, 0}},\
{GP_ELEMENT_BTN_BUTTON, { 18, 26,  27, 36, 1, 1,    GAMEPAD_MASK_B2, GP_SHAPE_SQUARE, 0, 0, 0}},\
{GP_ELEMENT_BTN_BUTTON, { 31, 26,  40, 36, 1, 1,    GAMEPAD_MASK_B4, GP_SHAPE_SQUARE, 0, 0, 0}},\
{GP_ELEMENT_BTN_BUTTON, { 44, 26,  53, 36, 1, 1,    GAMEPAD_MASK_B3, GP_SHAPE_SQUARE, 0, 0, 0}},\
{GP_ELEMENT_BTN_BUTTON, { 57, 26,  66, 36, 1, 1,    GAMEPAD_MASK_L1, GP_SHAPE_SQUARE, 0, 0, 0}},\

This is the original guitar hero and rock band guitar button layout. Thanks for reading and I hope someone can change this without it being a big hassle.

mikepparks, 2025-06-21:
If you do make a fork of the repo, you could always submit that as a PR and we can merge it in.

TheTrainGoes, 2025-07-02:
I've made a PR for this change, closing issue.

closed 2025-07-02 by TheTrainGoes