OpenStickCommunity/GP2040-CE #1422: Macro doesn't do anything

PhantomOSTeam, 2025-07-10:
### Expected Behavior

I have enabled macro and set it up correctly using XInput "LT". I'm expecting that after pressing LT macro will start to work. 

### Current Behavior

It doesn't do anything I tried to setup different buttons map but still absolutely nothing occurs.

### Context

[image]
[image]

TheTrainGoes, 2025-07-10:
Hi @PhantomOSTeam ,

This macro does not look like it is setup correctly.  

Please give a read over https://gp2040-ce.info/web-configurator/menu-pages/macros for more information on how to setup macros.  

You have told it to use a macro button + an input button but it does not sound like you are using a combination of buttons to make this work.  

Also, you cannot have a 0 frame input.

Lastly, we do not condone the use of macros for doing series of inputs in fighting games.  It is recommended to practice the inputs so you can do them on your own.

PhantomOSTeam, 2025-07-10:
I replicate  from macros documentation it still doesn't do anything... None of the button make any macro move. Am I missing something then? I also tried to setup any button to trigger and also without none of them is working

TheTrainGoes, 2025-07-10:
Hi @PhantomOSTeam ,

Again, I don't think you are setting the macro up properly.  

If you want a single button to activate the macro you will need to set it in macro without the macro + button input I talked about above and then assign the macro number in the GPIO mapping page directly.

Again, we do not condone the use of macros for doing series of inputs in fighting games. It is recommended to practice the inputs so you can do them on your own.

closed 2025-07-11 by TheTrainGoes