OpenStickCommunity/GP2040-CE #1456: mbedtls build error

Poldraunic, 2025-08-14:
I get a bunch of errors building latest master.

<details>

```
In file included from /home/joe/programming/GP2040-CE/build/_deps/pico_sdk-src/lib/mbedtls/include/mbedtls/build_info.h:192,
                 from /home/joe/programming/GP2040-CE/build/_deps/pico_sdk-src/lib/mbedtls/include/mbedtls/rsa.h:20,
                 from /home/joe/programming/GP2040-CE/headers/drivers/ps4/PS4Auth.h:5,
                 from /home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:4:
/home/joe/programming/GP2040-CE/build/_deps/pico_sdk-src/lib/mbedtls/include/mbedtls/check_config.h:748:2: error: #error "MBEDTLS_RSA_C defined, but not all prerequisites"
  748 | #error "MBEDTLS_RSA_C defined, but not all prerequisites"
      |  ^~~~~
/home/joe/programming/GP2040-CE/build/_deps/pico_sdk-src/lib/mbedtls/include/mbedtls/check_config.h:876:2: error: #error "MBEDTLS_SSL_ASYNC_PRIVATE defined, but not all prerequisites"
  876 | #error "MBEDTLS_SSL_ASYNC_PRIVATE defined, but not all prerequisites"
      |  ^~~~~
/home/joe/programming/GP2040-CE/build/_deps/pico_sdk-src/lib/mbedtls/include/mbedtls/check_config.h:1065:2: error: #error "MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH defined, but not all prerequisites"
 1065 | #error "MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH defined, but not all prerequisites"
      |  ^~~~~
/home/joe/programming/GP2040-CE/build/_deps/pico_sdk-src/lib/mbedtls/include/mbedtls/check_config.h:1110:2: error: #error "MBEDTLS_CHECK_PARAMS was removed in Mbed TLS 3.0. See https://github.com/Mbed-TLS/mbedtls/issues/4313"
 1110 | #error "MBEDTLS_CHECK_PARAMS was removed in Mbed TLS 3.0. See https://github.com/Mbed-TLS/mbedtls/issues/4313"
      |  ^~~~~
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp: In member function 'void PS4Auth::keyModeInitialize()':
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:19:85: error: 'mbedtls_mpi' has no non-static data member named 's'
   19 |     mbedtls_mpi name = { .s=1, .n=size / sizeof(mbedtls_mpi_uint), .p=bytes ## name }; \
      |                                                                                     ^
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:77:5: note: in expansion of macro 'NEW_CONFIG_MPI'
   77 |     NEW_CONFIG_MPI(N, options.rsaN.bytes, options.rsaN.size)
      |     ^~~~~~~~~~~~~~
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:19:85: error: 'mbedtls_mpi' has no non-static data member named 's'
   19 |     mbedtls_mpi name = { .s=1, .n=size / sizeof(mbedtls_mpi_uint), .p=bytes ## name }; \
      |                                                                                     ^
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:78:5: note: in expansion of macro 'NEW_CONFIG_MPI'
   78 |     NEW_CONFIG_MPI(E, options.rsaE.bytes, options.rsaE.size)
      |     ^~~~~~~~~~~~~~
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:19:85: error: 'mbedtls_mpi' has no non-static data member named 's'
   19 |     mbedtls_mpi name = { .s=1, .n=size / sizeof(mbedtls_mpi_uint), .p=bytes ## name }; \
      |                                                                                     ^
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:79:5: note: in expansion of macro 'NEW_CONFIG_MPI'
   79 |     NEW_CONFIG_MPI(P, options.rsaP.bytes, options.rsaP.size)
      |     ^~~~~~~~~~~~~~
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:19:85: error: 'mbedtls_mpi' has no non-static data member named 's'
   19 |     mbedtls_mpi name = { .s=1, .n=size / sizeof(mbedtls_mpi_uint), .p=bytes ## name }; \
      |                                                                                     ^
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:80:5: note: in expansion of macro 'NEW_CONFIG_MPI'
   80 |     NEW_CONFIG_MPI(Q, options.rsaQ.bytes, options.rsaQ.size)
      |     ^~~~~~~~~~~~~~
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:81:21: error: too many arguments to function 'void mbedtls_rsa_init(mbedtls_rsa_context*)'
   81 |     mbedtls_rsa_init(&ps4AuthData.rsa_context, MBEDTLS_RSA_PKCS_V21, MBEDTLS_MD_SHA256);
      |     ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/joe/programming/GP2040-CE/build/_deps/pico_sdk-src/lib/mbedtls/include/mbedtls/rsa.h:139:6: note: declared here
  139 | void mbedtls_rsa_init(mbedtls_rsa_context *ctx);
      |      ^~~~~~~~~~~~~~~~
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp: In member function 'void PS4Auth::keyModeProcess()':
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:108:14: error: 'mbedtls_sha256_ret' was not declared in this scope; did you mean 'mbedtls_sha256_free'?
  108 |         if ( mbedtls_sha256_ret(ps4AuthData.ps4_auth_buffer, 256, hashed_nonce, 0) < 0 ) {
      |              ^~~~~~~~~~~~~~~~~~
      |              mbedtls_sha256_free
/home/joe/programming/GP2040-CE/src/drivers/ps4/PS4Auth.cpp:112:17: error: 'MBEDTLS_RSA_PRIVATE' was not declared in this scope; did you mean 'MBEDTLS_PRIVATE'?
  112 |                 MBEDTLS_RSA_PRIVATE, MBEDTLS_MD_SHA256,
      |                 ^~~~~~~~~~~~~~~~~~~
      |                 MBEDTLS_PRIVATE
[1060/1410] Building CXX object CMakeFiles/GP2040-CE.dir/src/interfaces/i2c/ssd1306/obd_ssd1306.cpp.o
In file included from /home/joe/programming/GP2040-CE/headers/display/GPGFX_core.h:5,
                 from /home/joe/programming/GP2040-CE/headers/display/GPGFX.h:7,
                 from /home/joe/programming/GP2040-CE/headers/interfaces/i2c/ssd1306/obd_ssd1306.h:4,
                 from /home/joe/programming/GP2040-CE/src/interfaces/i2c/ssd1306/obd_ssd1306.cpp:1:
/home/joe/programming/GP2040-CE/headers/interfaces/i2c/displaybase.h: In member function 'virtual uint32_t GPGFX_DisplayBase::getPixel(uint8_t, uint8_t)':
/home/joe/programming/GP2040-CE/headers/interfaces/i2c/displaybase.h:21:58: warning: no return statement in function returning non-void [-Wreturn-type]
   21 |         virtual uint32_t getPixel(uint8_t x, uint8_t y) {}
      |                                                          ^
[... 249 lines cut ...]
      | 
In file included from /home/joe/programming/GP2040-CE/headers/display/GPGFX_core.h:5:
/home/joe/programming/GP2040-CE/headers/interfaces/i2c/displaybase.h: In member function 'virtual uint32_t GPGFX_DisplayBase::getPixel(uint8_t, uint8_t)':
/home/joe/programming/GP2040-CE/headers/interfaces/i2c/displaybase.h:21:58: warning: no return statement in function returning non-void [-Wreturn-type]
   21 |         virtual uint32_t getPixel(uint8_t x, uint8_t y) {}
      |                                                          ^
/home/joe/programming/GP2040-CE/src/display/ui/elements/GPMenu.cpp: In member function 'virtual void GPMenu::draw()':
/home/joe/programming/GP2040-CE/src/display/ui/elements/GPMenu.cpp:7:18: warning: unused variable 'baseX' [-Wunused-variable]
    7 |         uint16_t baseX = this->x;
      |                  ^~~~~
/home/joe/programming/GP2040-CE/src/display/ui/elements/GPMenu.cpp:8:18: warning: unused variable 'baseY' [-Wunused-variable]
    8 |         uint16_t baseY = this->y;
      |                  ^~~~~
/home/joe/programming/GP2040-CE/src/display/ui/elements/GPMenu.cpp:10:18: warning: unused variable 'menuWidth' [-Wunused-variable]
   10 |         uint16_t menuWidth = this->menuSizeX * 6;
      |                  ^~~~~~~~~
/home/joe/programming/GP2040-CE/src/display/ui/elements/GPMenu.cpp:11:18: warning: unused variable 'menuHeight' [-Wunused-variable]
   11 |         uint16_t menuHeight = this->menuSizeY * 8;
      |                  ^~~~~~~~~~
[1073/1410] Building CXX object CMakeFiles/GP2040-CE.dir/src/gp2040.cpp.o

```
</details>

Any tips?

mikepparks, 2025-08-16:
Which version of the Pico SDK?

Poldraunic, 2025-08-17:
@mikepparks, not sure. I "used" whatever was downloaded during CMake run.

Patataman, 2025-09-15:
@mikepparks I'm having the same problem, so I can give you the info:
- GP2040 v0.7.12-Beta2
- pico-sdk 2.2.0

On the other hand, using GP2040 v0.7.12-Beta2 or GP2040 v0.7.11 with pico-sdk 2.1.1 or 2.1.1-correct-picotool

<details><summary>Details</summary>

```
En el fichero incluido desde /home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/gen/parser.hpp:52,
                 desde /home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/pio_assembler.h:11,
                 desde /home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/main.cpp:8:
/home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/pio_types.h:290:5: error: ‘uint8_t’ no nombra a un tipo
  290 |     uint8_t used_gpio_ranges = 0;
      |     ^~~~~~~
/home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/pio_types.h:19:1: nota: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
   18 | #include "pio_enums.h"
  +++ |+#include <cstdint>
   19 | 
En el fichero incluido desde /home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/pio_assembler.h:12:
/home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/output_format.h:83:9: error: ‘uint8_t’ no nombra a un tipo
   83 |         uint8_t used_gpio_ranges;
      |         ^~~~~~~
/home/daniel/Documentos/taiko/pico-sdk/tools/pioasm/output_format.h:1:1: nota: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
  +++ |+#include <cstdint>
    1 | /*
[  2%] Building C object lib/SNESpad/CMakeFiles/SNESpad.dir/home/daniel/Documentos/taiko/pico-sdk/src/common/pico_sync/lock_core.c.o
[  2%] Building C object lib/SNESpad/CMakeFiles/SNESpad.dir/home/daniel/Documentos/taiko/pico-sdk/src/common/pico_sync/mutex.c.o
[  2%] Building C object lib/SNESpad/CMakeFiles/SNESpad.dir/home/daniel/Documentos/taiko/pico-sdk/src/common/pico_sync/critical_section.c.o
make[5]: *** [CMakeFiles/pioasm.dir/build.make:75: CMakeFiles/pioasm.dir/main.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:87: CMakeFiles/pioasm.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [pico-sdk/src/rp2_common/tinyusb/CMakeFiles/pioasmBuild.dir/build.make:95: pico-sdk/src/rp2_common/tinyusb/pioasm/src/pioasmBuild-stamp/pioasmBuild-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:2762: pico-sdk/src/rp2_common/tinyusb/CMakeFiles/pioasmBuild.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
```

</details> 

In all the cases I didn't have any difference from compiling picotool (2.2.0) along GP2040, or having it already installed.

I'm compiling in Manjaro, but afaik I have all the dependencies installed and compiling pico-sdk independently does not returns any error

Patataman, 2025-09-18:
I managed to compile (idk if it works correctly) but the main problems seems to be that pico-sdk uses the latest version of mbedTLS which is the 3.x and GP2040 is still using the functions and headers in 2.x

I manually went to the pico-sdk folder and did a checkout to mbedtls-2.28.10 in the mbedtls submodule and the compilation worked

Plaque-fcc, 2025-12-12:
Does this project reference that dependency version anywhere?